chipsalliance/rocket-chip #706: DefaultRV32Config won't build

mwachs5, 2017-04-25:
I don't care too much about `DefaultRV32Config`, but it seems like configs which are included should at least build or be removed:

```
Caused by: java.lang.IllegalArgumentException: requirement failed: The literal value 0 was elaborated with a specified width of 0 bits, but at least 1 bits are required.
    at scala.Predef$.require(Predef.scala:219)
    at chisel3.internal.firrtl.LitArg.<init>(IR.scala:68)
    at chisel3.internal.firrtl.ULit.<init>(IR.scala:77)
    at chisel3.core.UIntFactory$class.Lit(Bits.scala:548)
    at chisel3.core.UInt$.Lit(Bits.scala:565)
    at chisel3.core.package$fromBigIntToLiteral.asUInt(package.scala:48)
    at Chisel.package$UIntFactory$class.apply(compatibility.scala:62)
    at Chisel.package$UInt$.apply(compatibility.scala:148)
    at util.package$SeqToAugmentedSeq$.apply$extension(Package.scala:21)
    at tile.HasFPUParameters$class.recode(FPU.scala:341)
    at tile.FPUModule.recode(FPU.scala:361)
    at tile.FPU$$anonfun$6.apply$mcV$sp(FPU.scala:608)
```

aswaterman, 2017-04-26:
I think the short-term solution is to fix DefaultRV32Config and the long-term solution is to get all theoretically supported configs into some form of regression testing.

aswaterman, 2017-04-26:
Fixed on my branch

commit 684fa207 (2017-04-26): Support indexing 1-entry Seqs

It's a zero-width wire special case.

Closes #706.
commit 46c63dd9 (2017-04-27): Support indexing 1-entry Seqs

It's a zero-width wire special case.

Closes #706.
commit 1b0a1df2 (2017-04-27): Support indexing 1-entry Seqs

It's a zero-width wire special case.

Closes #706.
commit ee6702e5 (2017-04-27): Support indexing 1-entry Seqs

It's a zero-width wire special case.

Closes #706.
closed 2017-04-26 by aswaterman