chipsalliance/rocket-chip #712: Multicore RocketChip booting Linux

poinu, 2017-04-28:
We have built a dual core RocketChip verilator emulator using the configuration _DualCoreConfig_ privided in the repository. The build passes all tests, but we are having a hard time booting the Linux kernel and getting it recognize both cores of the RocketChip. The kernel has been configured with SMP option enabled (set to 8 CPUs, the default).

We are able to boot the kernel image on `spike`, simulating 4 processors with the option `-p`, and the kernel recognizes all 4 cores. However, when run in the _DualCoreConfig_ verilator emulator, only one core is recognized.

Our built is based in older priv-1.9 (I believe) commits as we haven't been able to boot the kernel with the latter priv-1.10 changes (not even when using _DefaultConfig_ nor spike).
- Commit rocket-chip: `f8bb67ab8fa625fc52fe221e140d469a64cda6a1`
- Commit riscv-tools: `886d8131dbd23533fb04d2d76a80be21d5f9ee7a` (tools embedded in rocket-chip repository)
- Commit riscv-linux: `030e8173dd9379de5c2f065ba73e734ff7647ef8`

Are we missing something? Any help is appreciated.

terpstra, 2017-07-17:
The bbl version in riscv-pk/priv-1.10 disables core 0. Is that maybe your problem?

poinu, 2017-07-18:
Hi @terpstra for you answer. To be honest I had completely forgotten about this issue... If I recall correctly, it had something to do with the bootrom used at that time. I think our build was based on priv-1.9, so I don't think the faced the bbl issue you describe, but thank you anyway :)

I'll just close the issue.