chipsalliance/rocket-chip #717: Tilelink and AXI

ViccoYang, 2017-05-02:
I studied this repo and the project-template. There is a MMIO  peripheral device example PWM in project template, but I saw both tilelink and AXI version in PWM.scala, so I'm confused the difference of these two version and I'm not sure where should the device be in this pic:
[image]
I think the AXI version of PWM should be at "AHB&APB Peripherals", but what about the tilelink version?
Thanks for your help!

terpstra, 2017-06-04:
There is no PWM in rocketchip? If you have a TL device, you probably want to connect it off the periphery bus, not the memory bus shown in your diagram.

muhammadali201, 2018-10-29:
is there any example of connecting AXI4 based peripherals with rocket-chip

sequencer, 2022-04-08:
See repo in https://github.com/sifive/sifive-blocks
But we need a test case to illustrate how to add new SoC devices.

closed 2022-04-08 by sequencer