chipsalliance/rocket-chip #733: Request to decouple PerfCounterIO's increment width from RetireWidth

ccelio, 2017-05-08:
Sorry for probably introducing the original hack, but I'd like to make a request to decouple the PerfCounterIO event's width from the retireWidth.

https://github.com/ucb-bar/rocket-chip/blob/master/src/main/scala/rocket/CSR.scala#L152

I would like a RocketCoreParams field that will let me choose what I want the event increment width to be.

My use-case is that a processor's issueWidth is likely to be wider than its retireWidth. Separating out different issue ports to separate events in not tenable, since software may want to save/restore counter state on a fine granularity. Even for coarse performance analysis on FPGAs, 29 hpm counters can run out fast.

Does this seem reasonable?

aswaterman, 2017-05-08:
Yes, but this interacts with how we're going to handle counters more systematically, so I'd like to have time to think about it more carefully than just adding another parameter to expand their width.

ccelio, 2017-05-08:
Sure thing, just want to make sure this is on the radar.

In the meantime, I'll continue with my fork.