chipsalliance/rocket-chip #736: Java error when make verilog with DefaultFPGAConfig

xinjic, 2017-05-10:
making verilog with DefaultConfig goes well, but working with DefaultFPGAConfig failed.

[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:236)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:234)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:234)
	at chisel3.Driver$.elaborate(Driver.scala:91)
Caused by: java.lang.IllegalArgumentException: requirement failed
	at scala.Predef$.require(Predef.scala:207)
	at chisel3.core.Vec$.do_apply(Aggregate.scala:70)
	at chisel3.core.Vec$.do_fill(Aggregate.scala:132)
	at uncore.devices.TLDebugModuleOuter$$anon$11.<init>(Debug.scala:345)
	at uncore.devices.TLDebugModuleOuter.module$lzycompute(Debug.scala:286)
	at uncore.devices.TLDebugModuleOuter.module(Debug.scala:286)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:60)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:60)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:60)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:57)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at diplomacy.LazyModule.instantiate(LazyModule.scala:57)
	at diplomacy.LazyModuleImp.<init>(LazyModule.scala:138)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 13 s, completed May 10, 2017 2:10:39 AM
make: *** [/home/jinxin/Projects/RISCV/rocket-chip/vsim/generated-src/rocketchip.DefaultFPGAConfig.fir] Error 1

Thanks!

sorear, 2017-05-10:
```
    Log fragments on GitHub are more legible if bracketed
    with ` characters as shown.
    ```

xinjic, 2017-05-10:
Thanks~
`[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
	at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
	at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
	at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
	at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
	at util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:44)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:91)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:236)
	at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:234)
	at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
	at chisel3.internal.Builder$.build(Builder.scala:234)
	at chisel3.Driver$.elaborate(Driver.scala:91)
Caused by: java.lang.IllegalArgumentException: requirement failed
	at scala.Predef$.require(Predef.scala:207)
	at chisel3.core.Vec$.do_apply(Aggregate.scala:70)
	at chisel3.core.Vec$.do_fill(Aggregate.scala:132)
	at uncore.devices.TLDebugModuleOuter$$anon$11.<init>(Debug.scala:345)
	at uncore.devices.TLDebugModuleOuter.module$lzycompute(Debug.scala:286)
	at uncore.devices.TLDebugModuleOuter.module(Debug.scala:286)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:60)
	at diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:60)
	at chisel3.core.Module$.do_apply(Module.scala:42)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:60)
	at diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:57)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at diplomacy.LazyModule.instantiate(LazyModule.scala:57)
	at diplomacy.LazyModuleImp.<init>(LazyModule.scala:138)
java.lang.RuntimeException: Nonzero exit code: 1
	at scala.sys.package$.error(package.scala:27)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
	at scala.Option.foreach(Option.scala:236)
	at sbt.BuildCommon$class.toError(Defaults.scala:2081)
	at sbt.Defaults$.toError(Defaults.scala:39)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
	at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
	at sbt.std.Transform$$anon$4.work(System.scala:63)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
	at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 15 s, completed May 10, 2017 3:13:06 AM
make: *** [/home/jinxin/Projects/RISCV/rocket-chip/vsim/generated-src/rocketchip.DefaultFPGAConfig.fir] Error 1
`

kllbrd, 2017-05-24:
@xinjic, try this instead:
make verilog config=DefaultFPGAConfig

stephengeerlings, 2017-09-25:
I have the same problem. 

@kllbrd your solution does not help, config is only set if you use uppercase.

I run:

`make verilog CONFIG=DefaultFPGAConfig`

My output:

'mkdir -p /mnt/c/dev/studie/home/rocket-chip/vsim/generated-src/
cd /mnt/c/dev/studie/home/rocket-chip && java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /mnt/c/dev/studie/home/rocket-chip/sbt-launch.jar "run-main freechips.rocketchip.system.Generator /mnt/c/dev/studie/home/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultFPGAConfig"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /mnt/c/dev/studie/home/rocket-chip/project
Using addons:
[info] Set current project to rocketchip (in build file:/mnt/c/dev/studie/home/rocket-chip/)
[info] Compiling 1 Scala source to /mnt/c/dev/studie/home/rocket-chip/chisel3/target/scala-2.11/classes...
[info] Packaging /mnt/c/dev/studie/home/rocket-chip/chisel3/target/scala-2.11/chisel3_2.11-3.1-SNAPSHOT.jar ...
[info] Done packaging.
[info] Running freechips.rocketchip.system.Generator /mnt/c/dev/studie/home/rocket-chip/vsim/generated-src freechips.rocketchip.system TestHarness freechips.rocketchip.system DefaultFPGAConfig
[info] [0.008] Elaborating design...
Interrupt map (0 harts 2 interrupts):
  [1, 2] => ExampleRocketSystem

[error] (run-main-0) java.lang.reflect.InvocationTargetException
java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:50)
        at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.internal.Builder$.build(Builder.scala:238)
        at chisel3.Driver$.elaborate(Driver.scala:92)
Caused by: java.lang.IllegalArgumentException: requirement failed
        at scala.Predef$.require(Predef.scala:207)
        at freechips.rocketchip.devices.tilelink.TLPLIC$$anon$1.<init>(Plic.scala:134)
        at freechips.rocketchip.devices.tilelink.TLPLIC.module$lzycompute(Plic.scala:111)
        at freechips.rocketchip.devices.tilelink.TLPLIC.module(Plic.scala:111)
        at freechips.rocketchip.devices.tilelink.TLPLIC.module(Plic.scala:64)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:61)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1$$anonfun$apply$5.apply(LazyModule.scala:61)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:61)
        at freechips.rocketchip.diplomacy.LazyModule$$anonfun$instantiate$1.apply(LazyModule.scala:58)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at freechips.rocketchip.diplomacy.LazyModule.instantiate(LazyModule.scala:58)
        at freechips.rocketchip.diplomacy.LazyMultiIOModuleImp.<init>(LazyModule.scala:162)
        at freechips.rocketchip.coreplex.BareCoreplexModule.<init>(BaseCoreplex.scala:24)
java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
        at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
        at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
        at scala.Option.foreach(Option.scala:236)
        at sbt.BuildCommon$class.toError(Defaults.scala:2081)
        at sbt.Defaults$.toError(Defaults.scala:39)
        at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
        at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
        at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
        at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
        at sbt.std.Transform$$anon$4.work(System.scala:63)
        at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
        at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
        at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)
[error] (rocketchip/compile:runMain) Nonzero exit code: 1
[error] Total time: 17 s, completed Sep 25, 2017 10:19:52 AM
/mnt/c/dev/studie/home/rocket-chip/vsim/Makefrag-verilog:12: recipe for target '/mnt/c/dev/studie/home/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultFPGAConfig.fir' failed
make: *** [/mnt/c/dev/studie/home/rocket-chip/vsim/generated-src/freechips.rocketchip.system.DefaultFPGAConfig.fir] Error 1'

PR #1047 (2017-10-11): system: fix DefaultFPGAConfig
It was missing cores. Fixes #736.
commit b566ffed (2017-10-11): system: fix DefaultFPGAConfig (#1047)

It was missing cores. Fixes #736.
closed 2017-10-11 by terpstra