chipsalliance/rocket-chip #737: How to mixin two trait with the same implicit val p: Parameters

colin4124, 2017-05-11:
I am working on Rocket Chip tag: 4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7 and Debug Spec. 0.11

I want to implement trait DebugModule as Hasti Slave. DebugModule inherits trait HasDebugModuleParameters which contains "implicit val p: Parameters", and HastiModule inherits trait HasHastiParameters which also contains **"implicit val p: Parameters"**.

Thus, I got an error:

    class HastiDebugModule inherits conflicting members:
    [error]   value p in class HastiModule of type config.Parameters  and
    [error]   value p in trait HasDebugModuleParameters of type config.Parameters

This should be scala problem, a class can't extends two traits both have the same variable name. However, the reason why I open this issue is that I want to know how to use the global  configuration implicit parameter **_p_**  in this case?

The trait DebugModule and trait HastiModule both have implicit parameter **_p_**  which can be to configure globally, should I design a class combine the two trait to use one **"implicit val p: Parameters"** in some way？ Or should I design a class just extends DebugModule and design another Module that converts the DebugModule's system bus IO interface to Hasti Slave IO interface?

@aswaterman Could you give me some advice?

megan-riscv-6thworkshop, 2017-05-11:
I am not sure exactly what you are trying to do, but if you just want the Debug Module to be an AXI slave instead of a TileLink slave, there is a simple way to do what you want:

https://github.com/ucb-bar/rocket-chip/blob/4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7/src/main/scala/uncore/devices/Debug.scala#L851

That line makes a "TLDebugModule" out of the internals of the Debug Module. think you can just do "AXIDebugModule" with that magic macro, and change 'TL' to 'AXI' to create an AXI slave Debug Module.

(Unfortunately, this trick doesn't work on master branch/Debug v13 because the magic macro isn't used)

megan-riscv-6thworkshop, 2017-05-11:
Ah actually maybe it needs to be 'AXI4'. See this code which is what does the magic:

https://github.com/ucb-bar/rocket-chip/blob/4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7/src/main/scala/uncore/axi4/RegisterRouter.scala

colin4124, 2017-05-11:
@megan-riscv-6thworkshop Thanks very much, I have a try.

megan-riscv-6thworkshop, 2017-05-11:
Or "AHB" is also a possibility.

colin4124, 2017-05-24:
@megan-riscv-6thworkshop @mwachs5 

The AXI4  implements does not has **`implicit val p`** , so I don't do it like AXI4. 

I do it directly, I remove **`implicit val p`** from HastiModule, and I try to modify trait **DebugModule** as a class, I mplement function `regmap` and `io` in **DebugModule**  itself. But I got error:

> error]  found   : chisel3.core.UInt
> [error]  required: Int
> [error]   dbRamRdData := dbRamRdDataFields(dbRamAddr)
> [error]                                    ^
> [error] one error found

I found [that](https://github.com/freechipsproject/rocket-chip/blob/4146f6a792277297bd80d4b6f1b6f2b3cb0f31d7/src/main/scala/uncore/devices/Debug.scala#L500) it declares `Wire(UInt(width=ramAddrWidth))` , but it is used as index(`Int` type) in `dbRamRdDataFields(dbRamAddr)`.

It is so confused me! Why when these codes were used as trait, there is no error, but when these codes are used as class, there is error as above.

colin4124, 2017-05-24:
I change `List` to `Vec` and then it is Ok.
```
val dbRamRdDataFields = List.tabulate(cfg.nDebugRamBytes / ramDataBytes) { ii =>
    val slice = ramMem.slice(ii * ramDataBytes, (ii+1)*ramDataBytes)
    slice.reduce[UInt]{ case (x: UInt, y: UInt) => Cat(y, x)}
}
```

```
val dbRamRdDataFields = Vec.tabulate(cfg.nDebugRamBytes / ramDataBytes) { ii =>
    val slice = ramMem.slice(ii * ramDataBytes, (ii+1)*ramDataBytes)
    slice.reduce[UInt]{ case (x: UInt, y: UInt) => Cat(y, x)}
}
```

mwachs5, 2017-05-24:
Ok. It would be good to understand what you are trying to do... why are you trying to put AXI between the Debug Module and the core?