chipsalliance/rocket-chip #743: Error while building the toolchain

kllbrd, 2017-05-16:
I have all the needed packets installed. I can't figure out what the problem is...

[root@localhost riscv-tools]# ./build.sh
Starting RISC-V Toolchain build process

libjaylink/Makefile.am:23: error: Libtool library used but 'LIBTOOL' is undefined
libjaylink/Makefile.am:23:   The usual way to define 'LIBTOOL' is to add 'LT_INIT'
libjaylink/Makefile.am:23:   to 'configure.ac' and run 'aclocal' and 'autoconf' again.
libjaylink/Makefile.am:23:   If 'LT_INIT' is in 'configure.ac', make sure
libjaylink/Makefile.am:23:   its definition is in aclocal's search path.
autoreconf: automake failed with exit status: 1

Thanks.

mwachs5, 2017-05-16:
Install libtool? I think since we added openocd we need to add more deps

sudo apt-get install libtool

kllbrd, 2017-05-16:
Thank you, but I have already done this and I am taking an error because of the version of libusb I think:
configure: error: Package requirements (libusb-1.0 >= 1.0.9) were not met:

No package 'libusb-1.0' found

Consider adjusting the PKG_CONFIG_PATH environment variable if you
installed software in a non-standard prefix.

Alternatively, you may set the environment variables libusb_CFLAGS
and libusb_LIBS to avoid the need to call pkg-config.
See the pkg-config man page for more details.
configure: error: ../../../../../src/jtag/drivers/libjaylink/configure.gnu failed for src/jtag/drivers/libjaylink

hellacod, 2017-05-18:
Hello, is there any solution for this error yet?
I have this problem on my fedora distribution and on my KDE.

kllbrd, 2017-05-19:
I found a solution for the libusb error, simply sudo yum install libusb-devel.
As for the rest errors about libtoolize I have no idea: 

libtoolize:   error: 'build-aux/config.guess' is newer: use '--force' to overwrite
libtoolize:   error: 'build-aux/config.sub' is newer: use '--force' to overwrite

I also take these after the libusb install:

../../../src/target/riscv/riscv-013.c: In function ‘assert_reset’:
../../../src/target/riscv/riscv-013.c:71:85: error: left shift of negative value [-Werror=shift-negative-value]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1164:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1164:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: left shift of negative value [-Werror=shift-negative-value]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1167:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 0);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1167:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 0);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c: In function ‘deassert_reset’:
../../../src/target/riscv/riscv-013.c:71:85: error: left shift of negative value [-Werror=shift-negative-value]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1193:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1193:15: note: in expansion of macro ‘set_field’
     control = set_field(control, DMI_DMCONTROL_HALTREQ, 1);
               ^~~~~~~~~
../../../src/target/riscv/riscv-013.c: In function ‘riscv013_halt_current_hart’:
../../../src/target/riscv/riscv-013.c:71:85: error: left shift of negative value [-Werror=shift-negative-value]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1703:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 1);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1703:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 1);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: left shift of negative value [-Werror=shift-negative-value]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
../../../src/target/riscv/riscv-013.c:1719:14: note: in expansion of macro ‘set_field’
  dmcontrol = set_field(dmcontrol, DMI_DMCONTROL_HALTREQ, 0);
              ^~~~~~~~~
../../../src/target/riscv/riscv-013.c:71:85: error: result of ‘-2147483648 << 1’ requires 33 bits to represent, but ‘int’ only has 32 bits [-Werror=shift-overflow=]
 #define set_field(reg, mask, val) (((reg) & ~(mask)) | (((val) * ((mask) & ~((mask) << 1))) & (mask)))
                                                                                     ^
[... cut ...]
   aa = set_field(aa, DMI_ABSTRACTAUTO_AUTOEXECPROGBUF, aa_aed);
        ^~~~~~~~~
cc1: all warnings being treated as errors
gmake[6]: *** [riscv/riscv-013.lo] Error 1
gmake[5]: *** [all-recursive] Error 1
gmake[4]: *** [all] Error 2
gmake[3]: *** [all-recursive] Error 1
gmake[2]: *** [all] Error 2
gmake[1]: *** [all-recursive] Error 1
gmake: *** [all] Error 2

I tried to bootstrap and then configure the openocd but I am taking these two again and again.
If I find anything new I'll post it. 
Any help would be appreciated!

yuz835, 2017-07-10:
have u fixed?

sequencer, 2022-04-08:
this depends on the new test framework. #2953 
basically I think we need to separate the build receipt from RC, I'm thinking using Nix, and providing a alternative way to let user be able to use the distro toolchains.

closed 2022-04-08 by sequencer