chipsalliance/rocket-chip #758: make run error in emulator

ninode, 2017-05-22:
Hello, 

I got these errors in emulator. I have never got them before. Could you please tell me why and how can I fix them ?

Thank you so much

```
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator$ make -j8 run-asm-tests
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64mi-p-ma_addr 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64mi-p-ma_addr.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64mi-p-ma_fetch 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64mi-p-ma_fetch.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64mi-p-sbreak 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64mi-p-sbreak.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64mi-p-scall 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64mi-p-scall.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64mi-p-breakpoint 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64mi-p-breakpoint.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64ui-p-simple 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64ui-p-simple.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64ui-p-add 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64ui-p-add.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/rv64ui-p-addi 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/rv64ui-p-addi.out && [ $PIPESTATUS -eq 0 ]
make: *** [output/rv64mi-p-breakpoint.out] Error 1
make: *** Waiting for unfinished jobs....
make: *** [output/rv64ui-p-addi.out] Error 1
make: *** [output/rv64mi-p-sbreak.out] Error 1
make: *** [output/rv64mi-p-scall.out] Error 1
make: *** [output/rv64mi-p-ma_fetch.out] Error 1
make: *** [output/rv64ui-p-simple.out] Error 1
make: *** [output/rv64mi-p-ma_addr.out] Error 1
make: *** [output/rv64ui-p-add.out] Error 1
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator$ make -j8 run-bmark-tests
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/mm.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/mm.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/spmv.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/spmv.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/mt-vvadd.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/mt-vvadd.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/median.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/median.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/multiply.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/multiply.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/qsort.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/qsort.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/towers.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/towers.riscv.out && [ $PIPESTATUS -eq 0 ]
./emulator-rocketchip-DefaultConfig +max-cycles=100000000 +verbose output/vvadd.riscv 3>&1 1>&2 2>&3 | /home/alpha/fpga-zynq/rocket-chip/riscv/bin/spike-dasm  > output/vvadd.riscv.out && [ $PIPESTATUS -eq 0 ]
make: *** [output/spmv.riscv.out] Error 1
make: *** Waiting for unfinished jobs....
make: *** [output/multiply.riscv.out] Error 1
make: *** [output/qsort.riscv.out] Error 1
make: *** [output/vvadd.riscv.out] Error 1
make: *** [output/mm.riscv.out] Error 1
make: *** [output/mt-vvadd.riscv.out] Error 1
make: *** [output/towers.riscv.out] Error 1
make: *** [output/median.riscv.out] Error 1

```

sorear, 2017-05-22:
Since error messages were redirected to `output/**.out`, maybe check there?

ninode, 2017-05-22:
sorear, Thank you. The generated .out test files have the following error. Do you have any idea where to go from here ? 

**emulator-rocketchip-DefaultConfig: ../fesvr/dtm.cc:67: uint32_t dtm_t::run_program(const uint32_t*, size_t, size_t): Assertion `n <= ram_words()' failed.**

mwachs5, 2017-05-22:
You probably need to update and re-compile your RISC-V tools probably.
FESVR changed along with a lot of other things.

ninode, 2017-05-24:
Thank you mwachs5. I updated the toolchain and re-compiled it. It worked.

closed 2017-05-24 by ninode

mysoreanoop, 2020-08-13:
@ninode 
I am facing the same error. I updated the riscv-tools and riscv-gnu-toolchain as well. It still throws the same error. However, it reads:
`emulator-freechips.rocketchip.system-freechips.rocketchip.system.DefaultConfig: ../fesvr/dtm.cc:61: uint32_t dtm_t::do_command(dtm_t::req): Assertion `resp_buf.resp == 0' failed.`

Could perhaps a newer version of riscv-tools not be working?

mysoreanoop, 2020-08-24:
Yes, I reinstalled the riscv-tools (riscv-isa-simulator) properly.

Rahul-Kande, 2020-08-24:
> Yes, I reinstalled the riscv-tools (riscv-isa-simulator) properly.

That worked for me too. Thank you. 

I think the issue was that I was using the older version of fesvr and it was resolved when I reinstalled using the riscv-isa-sim which has the latest fesvr in it.