chipsalliance/rocket-chip #763: generated C++ code in emulator/generated-src

ninode, 2017-05-24:
Hello,

In the emulator/generated-src directory, I didn't find the Chisel generated C++ code, like I'm supposed to in the instructions.

Here's what I find: 
```
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator$ cd generated-src
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator/generated-src$ ls
DefaultConfig.dts      rocketchip.DefaultConfig      rocketchip.DefaultConfig.v
DefaultConfig.graphml  rocketchip.DefaultConfig.d
DefaultConfig.json     rocketchip.DefaultConfig.fir

alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator/generated-src$ cd rocketchip.DefaultConfig/
alpha@alpha-VirtualBox:~/fpga-zynq/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig$ ls
emulator.d                        VTestHarness__5__Slow.cpp
emulator.o                        VTestHarness__5__Slow.d
SimDTM.d                          VTestHarness__5__Slow.o
SimDTM.o                          VTestHarness__6.cpp
verilated.d                       VTestHarness__6.d
verilated_dpi.d                   VTestHarness__6.o
verilated_dpi.o                   VTestHarness__6__Slow.cpp
verilated.o                       VTestHarness__6__Slow.d
VTestHarness___024unit.cpp        VTestHarness__6__Slow.o
VTestHarness___024unit.d          VTestHarness__7.cpp
VTestHarness___024unit.h          VTestHarness__7.d
VTestHarness___024unit.o          VTestHarness__7.o
VTestHarness___024unit__Slow.cpp  VTestHarness__7__Slow.cpp
VTestHarness___024unit__Slow.d    VTestHarness__7__Slow.d
VTestHarness___024unit__Slow.o    VTestHarness__7__Slow.o
VTestHarness__1.cpp               VTestHarness__8.cpp
VTestHarness__1.d                 VTestHarness__8.d
VTestHarness__1.o                 VTestHarness__8.o
VTestHarness__1__Slow.cpp         VTestHarness__9.cpp
VTestHarness__1__Slow.d           VTestHarness__9.d
VTestHarness__1__Slow.o           VTestHarness__9.o
VTestHarness__2.cpp               VTestHarness__ALL.a
VTestHarness__2.d                 VTestHarness_classes.mk
VTestHarness__2.o                 VTestHarness.cpp
VTestHarness__2__Slow.cpp         VTestHarness.d
VTestHarness__2__Slow.d           VTestHarness__Dpi.cpp
VTestHarness__2__Slow.o           VTestHarness__Dpi.d
VTestHarness__3.cpp               VTestHarness__Dpi.h
VTestHarness__3.d                 VTestHarness__Dpi.o
VTestHarness__3.o                 VTestHarness.h
VTestHarness__3__Slow.cpp         VTestHarness__Inlines.h
VTestHarness__3__Slow.d           VTestHarness.mk
VTestHarness__3__Slow.o           VTestHarness.o
VTestHarness__4.cpp               VTestHarness__Slow.cpp
VTestHarness__4.d                 VTestHarness__Slow.d
VTestHarness__4.o                 VTestHarness__Slow.o
VTestHarness__4__Slow.cpp         VTestHarness__Syms.cpp
VTestHarness__4__Slow.d           VTestHarness__Syms.d
VTestHarness__4__Slow.o           VTestHarness__Syms.h
VTestHarness__5.cpp               VTestHarness__Syms.o
VTestHarness__5.d                 VTestHarness__ver.d
VTestHarness__5.o                 VTestHarness__verFiles.dat
```
Am I doing something wrong ? 

Thank you

colinschmidt, 2017-05-24:
The generated C++ being inside another directory is expected, we need to update that section of the instructions.

For the second question: the vsim directory requires vcs, and that error will stop any simulator from happening. You can simulate rocket-chip in the emulator directory without access to that tool.

You mention an error in the emulator but I don't believe you've posted it (unless you mean where the C++ files are).

ninode, 2017-06-07:
The confusion about the emulator is about the location of the C++ files. 
This answers my question.

Thank you very much

closed 2017-06-07 by ninode