chipsalliance/rocket-chip #764: VCS doesn't compile when TLMonitorBuilder is set to None

aswaterman, 2017-05-25:
...presumably because there are no instantiations of plusarg_reader, causing VCS to instantiate one as a top-level module.  I'm currently working around this by changing the "borked" string to "%d".

`cd . && ./simv-rocketchip-DefaultConfig-debug -q +ntb_random_seed_automatic +verbose +vcdplusfile=output/rv64ud-p-fdiv.vpd +max-cycles=100000000 output/rv64ud-p-fdiv 3>&1 1>&2 2>&3 | /scratch/waterman/toolchain/bin/spike-dasm  > output/rv64ud-p-fdiv.out && [ $PIPESTATUS -eq 0 ]

Note-[RT_BS] Larger stack needed
  Note: Bumping stack limit from 8192 to 8193 Kbytes.

NOTE: automatic random seed used: 2417681561

Error-[STASKE_VPAMHFS] Invalid argument to $value$plusargs
/scratch/waterman/federation/rocket-chip/vsrc/plusarg_reader.v, 17
  An invalid plusarg_format string 'borked' is passed as first argument to 
  $value$plusargs.
  Please ensure that the first argument to $value$plusargs is a valid 
  plusarg_format specifier string.
`

aswaterman, 2017-05-25:
If you don't object, I can commit my workaround.

terpstra, 2017-05-25:
Lol. That's a funny bug. Your fix seems fine to me.

terpstra, 2017-05-25:
Actually, use "borked=%d"

aswaterman, 2017-05-25:
ok :)

PR #765 (2017-05-25): Make plusarg_reader default args work with VCS
Resolves #764
commit 618468a0 (2017-05-25): Make plusarg_reader default args work with VCS (#765)

Resolves #764
closed 2017-05-25 by aswaterman