chipsalliance/rocket-chip #774: Should the default JTAG IDCODE be 0x1?

edwardcwang, 2017-06-01:
https://github.com/freechipsproject/rocket-chip/blob/9002e7e53299598027f41b95dba1666310923699/src/main/scala/rocketchip/DebugTransport.scala#L22

```scala
class JtagDTMKeyDefault extends JtagDTMConfig(
  idcodeVersion = 0,
  idcodePartNum = 0,
  idcodeManufId = 0,
debugIdleCycles = 5) // Reasonable guess for synchronization.
```

Should the default IDCODE be 0x1, or the same as in the previous debug module (0x10e31913)? Or should it throw an error message to prevent rocket from elaborating if the JTAG is enabled but the idcode is not specified?

@mwachs5 @palmer-dabbelt @ducky64

mwachs5, 2017-06-01:
The IDCODE (0x10e31913) uses SiFive's assigned JEDEC IDCODE. So in general those who aren't SiFive should not be using that IDCODE. That's why we removed it from the default.

Those using rocket-chip repo should probably override this default value to something meaningful for their design. The ManufId is actually exported from the ExampleTop as part of its I/O and is tied off in the TestHarness.

edwardcwang, 2017-06-01:
I see, that makes sense. Thanks for the explanation. Would having an elaboration-time warning for users of rocket-chip be useful so that they don't run into surprises when they try to program their cores?

mwachs5, 2017-06-01:
When you say "program their cores" do you mean when using OpenOCD to communicate with them? Is there actually a value that you use there that causes a problem? riscv-openocd is very tolerant of the IDCODE, it may print a warning if it's not the expected ID.

I'm not sure what the elaboration time warning would be because the mfr_id could go all the way out the pins of the chip. I added a runtime warning (so if you ran with simulate it could at least detect illegal idcodes, though the chances you detect an unconnected mfr_id are somewhat low):

https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/jtag/JtagTap.scala#L175

mwachs5, 2017-06-01:
NOte that if you did actually use the defaults here, then your IDCODE would be 0x1, because there is a bit in the IDCODE "always 1":
https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocketchip/DebugTransport.scala#L230

edwardcwang, 2017-06-02:
@mwachs5 Ah, I see. I think I see where my confusion was coming from now.

In https://github.com/freechipsproject/rocket-chip/blob/9002e7e53299598027f41b95dba1666310923699/src/main/scala/rocketchip/DebugTransport.scala#L231 it uses two out of the three `Int`s in `JtagDTMConfig` but doesn't actually 
use the third one. I was confused since I thought that all 3 would be determined at elaboration-time by `JtagDTMConfig`. In my design, I think I left io.jtag_mfr_id unconnected and the synthesis tool tied it at some point to zero.

Perhaps if we are not actually going to use `c.idcodeManufId.U`, perhaps it could be a good idea to put in a comment saying "Please wire up io.jtag_mfr_id to set the mfr_id or something"? The TestHarness I think manually goes and re-retrieves it (or maybe just remove `idcodeManufId` from `JtagDTMConfig` to avoid confusion perhaps).

```scala
  idcode.always1    := 1.U
  idcode.version    := c.idcodeVersion.U
  idcode.partNumber := c.idcodePartNum.U
  idcode.mfrId := io.jtag_mfr_id
```

mwachs5, 2017-06-02:
The comment seems like a reasonable request. Keep in mind that RocketChipExampleTop is just an example top level, and other designs could actually wire up mfrId using the value in JtagDTMConfig,  I think we should leave it there but comment it as you suggest.

edwardcwang, 2017-06-02:
Thoughts on https://github.com/freechipsproject/rocket-chip/pull/778 ?

PR #778 (2017-06-03): Add a note to wire up jtag_mfr_id
Close #774
commit cdbf67be (2017-06-03): Add a note to wire up jtag_mfr_id (#778)

Close #774
closed 2017-06-03 by terpstra