chipsalliance/rocket-chip #780: %Error: Verilator threw signal 9.  Consider trying --debug --gdbbt

promach, 2017-06-04:
I have error: Verilator threw signal 9.  Consider trying --debug --gdbbt   when I try using  

> make -j8 run-asm-tests

in the emulator directory

mkdir -p /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig
/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/promach/Documents/Grive/Personal/CPU/RISC-V/toolchain/include -I/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc/verilator.h" -Mdir /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig \
-o /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc/emulator.cc /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/promach/Documents/Grive/Personal/CPU/RISC-V/toolchain/lib -Wl,-rpath,/home/promach/Documents/Grive/Personal/CPU/RISC-V/toolchain/lib -L/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator -lfesvr -lpthread" \
-CFLAGS "-I/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src -include /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h"
%Error: Verilator threw signal 9.  Consider trying --debug --gdbbt
%Error: Command Failed /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/home/promach/Documents/Grive/Personal/CPU/RISC-V/toolchain/include -I/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc/verilator.h' -Mdir /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig -o /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/emulator-rocketchip-DefaultConfig /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig.v /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc/emulator.cc /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/home/promach/Documents/Grive/Personal/CPU/RISC-V/toolchain/lib -Wl,-rpath,/home/promach/Documents/Grive/Personal/CPU/RISC-V/toolchain/lib -L/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src -include /home/promach/Documents/Grive/Personal/CPU/RISC-V/rocket-chip/emulator/generated-src/rocketchip.DefaultConfig/VTestHarness.h'
make: *** [Makefrag-verilator:69: emulator-rocketchip-DefaultConfig] Error 255

[error.txt](https://github.com/freechipsproject/rocket-chip/files/1050320/error.txt)

terpstra, 2017-06-04:
Are you using the newest Verilator? We started using a new plusargs feature recently.

promach, 2017-06-05:
@terpstra I am using Verilator 3.886 2016-07-30 rev verilator_3_884-13-gb4a7e46

What do you mean by new plusargs feature ?
Could you advice how I should modify the code to suit the new verilator ?

terpstra, 2017-06-05:
You need 3.904 or newer.

promach, 2017-06-06:
@terpstra Thanks. This particular error is solved

closed 2017-06-06 by promach

JunningWu, 2018-04-13:
I have the same error.
although I have changed verilator version from 3.884 all the way to 3.904/3.922, did not resolve the error.

env info:
ubuntu 16.04 VMware
latest git code
 

```
Verilator 3.922 2018-03-17 rev verilator_3_920-32-gdf3d1a4
```

```
[info] [34.678] Done elaborating.
[success] Total time: 55 s, completed Apr 13, 2018 9:48:16 AM
mkdir -p /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/riscv/RISC-V/riscv/freedom/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.fir -o /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v -X verilog
Java HotSpot(TM) 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
mkdir -p /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig
/home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/riscv-tools/riscv-tc/include -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc -DVERILATOR -include /home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc/verilator.h" -Mdir /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig \
	-o /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc/emulator.cc /home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/riscv/RISC-V/riscv/freedom/rocket-chip/riscv-tools/riscv-tc/lib -Wl,-rpath,/home/riscv/RISC-V/riscv/freedom/rocket-chip/riscv-tools/riscv-tc/lib -L/home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator -lfesvr -lpthread" \
	-CFLAGS "-I/home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src -include /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h"
%Error: Verilator threw signal 9.  Consider trying --debug --gdbbt
%Error: Command Failed /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/verilator/install/bin/verilator_bin --cc --exe --top-module TestHarness \+define\+PRINTF_COND\=\$c\(\"verbose\"\,\"\&\&\"\,\"done_reset\"\) \+define\+STOP_COND\=\$c\(\"done_reset\"\) --assert --output-split 20000 -Wno-STMTDLY --x-assign unique -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/vsrc -O3 -CFLAGS -O1\ -std\=c\+\+11\ -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/riscv-tools/riscv-tc/include\ -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc\ -DVERILATOR\ -include\ /home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc/verilator.h -Mdir /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig -o /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/emulator-rocketchip-BOOMConfig /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig.v /home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc/emulator.cc /home/riscv/RISC-V/riscv/freedom/rocket-chip/csrc/SimDTM.cc -LDFLAGS \ -L/home/riscv/RISC-V/riscv/freedom/rocket-chip/riscv-tools/riscv-tc/lib\ -Wl\,-rpath\,/home/riscv/RISC-V/riscv/freedom/rocket-chip/riscv-tools/riscv-tc/lib\ -L/home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator\ -lfesvr\ -lpthread -CFLAGS -I/home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src\ -include\ /home/riscv/RISC-V/riscv/freedom/rocket-chip/emulator/generated-src/rocketchip.BOOMConfig/VTestHarness.h
make: *** [emulator-rocketchip-BOOMConfig] Error 255
```

JunningWu, 2018-04-13:
@terpstra @promach