chipsalliance/rocket-chip #793: Fails during sub module init

pkimelman-nxp, 2017-06-09:
Following the directions on MacOS (El Capitan), the submodule init fails on riscv-tools:
error: no such remote ref 21113949ff22bd45c6d5928f7b6214c7e0a88bde
Fetched in submodule path 'riscv-tools', but it did not contain 21113949ff22bd45c6d5928f7b6214c7e0a88bde. Direct fetching of that commit failed.

Not sure what causes and tried an update but it says it is up to date.

aswaterman, 2017-06-09:
Seems we screwed up and pointed rocket-chip to a riscv-tools pointer that wasn't properly merged into the master branch.  This will be fixed shortly.

aswaterman, 2017-06-09:
@pkimelman-nxp pull from master, and things should be in order once again.  Sorry about that.

closed 2017-06-09 by aswaterman