chipsalliance/rocket-chip #809: rocc inst encountered an illegal instrution error

camelolim, 2017-06-20:
We realized a rocc extension instruction. When called by user application, we encountered an illegal instruction error.

We found the error comes from decode unit in rocket.scala, as below
```
  val id_illegal_insn = !id_ctrl.legal ||
    id_ctrl.div && !csr.io.status.isa('m'-'a') ||
    id_ctrl.amo && !csr.io.status.isa('a'-'a') ||
    id_ctrl.fp && !(csr.io.status.fs.orR && csr.io.status.isa('f'-'a')) ||
    id_ctrl.dp && !csr.io.status.isa('d'-'a') ||
    ibuf.io.inst(0).bits.rvc && !csr.io.status.isa('c'-'a') ||
    id_ctrl.rocc && !(csr.io.status.xs.orR && csr.io.status.isa('x'-'a'))
```

And it refers to status.xs, as it is set to 0 when rocc is called.
My question is, how to set it? By hardware(when extension instructon is called) or by software(in user application, before extension instruction is called)?

camelolim, 2017-06-20:
I run the instruction in emulator, the result is correct, and have no illegal instruction error.

sorear, 2017-06-20:
If you use RoCC, you need to modify riscv-pk to enable RoCC when running your test.

See https://groups.google.com/a/groups.riscv.org/d/msg/sw-dev/TPWqS7x_0y0/xs51tk45FwAJ

camelolim, 2017-06-21:
@sorear Thank you so much. I know the reason. Let me have a try.

closed 2017-06-21 by camelolim