chipsalliance/rocket-chip #817: RegisterReadIO and RegisterWriteIO need cloneType method

zhemao, 2017-06-24:
The regmapper interface is pretty slick, but one hangup is that RegisterReadIO and RegisterWriteIO don't have cloneType methods. That means you can't wrap them in a Wire, which would be very useful if you just want to connect existing decoupled interfaces to request and response. Right now, I work around it using the function-based constructor, but it would be more convenient to just declare a wire.

zhemao, 2017-07-10:
@terpstra could you take a look at this? I can submit a pull request if need be, but the change required should be pretty straightforward.

terpstra, 2017-07-17:
RefReadFn and RegWriteFn already accept a ReadyValidIO, from which DecoupledIO is derived. That said, if you want to add a cloneType, I'd merge it.

zhemao, 2017-07-18:
Yes, but using a ReadyValidIO is semantically different from using a RegisterReadIO. It doesn't allow you to decoupled the read request from the read response. I've submitted a PR.

PR #874 (2017-07-19): Add cloneType to RegisterWriteIO and RegisterReadIO
This gives a cloneType to RegisterWriteIO and RegisterReadIO so that it can be used in Wire definitions. Fixes #817.
closed 2017-07-19 by terpstra