chipsalliance/rocket-chip #831: Make it so we don't need to do the travis hackery ever again

mwachs5, 2017-06-29:
branches:
  only:
    - "master"

Apparently we can add that to our travis.yml and then we dont need to keep switching the "build branch updates" toggle:

(from this thread: https://github.com/travis-ci/travis-ci/issues/1147)

terpstra, 2017-06-29:
You would still end up having to run travis twice for every PR. Once in push and once on merge. Given the limited build power, I'm not sure this is a good plan.

mwachs5, 2017-06-29:
You could always kill the merge job... Now that you are allowing merge commits which are true merges, I don't think that's really a bad idea.

terpstra, 2017-06-29:
If we had infinite build jobs, I would totally be in favour of this PR. I suppose it won't bother anyone with Travis admin privileges, because they can cancel it, as you suggest. I dunno. I abstain from reviewing this PR. Let's see what @aswaterman and @hcook think.

mwachs5, 2017-06-29:
Yeah. I think the other right fix is to find a way to make the riscv-tools bump less painful. Anyway I don't feel too strongly about it, just googling while waiting for runs to pass :)

closed 2017-07-05 by mwachs5