chipsalliance/rocket-chip #833: How to get riscv64-unknown-elf-gcc to see installed nonstandard library?

NewPath2015, 2017-06-29:
Hi I followed the instructions in this [issue](https://github.com/freechipsproject/rocket-chip/issues/667)  to run the hello world program on the generated emulator. I wrote a simple C program that uses [fftw](http://www.fftw.org/) which I have installed. It compiles and runs fine using this command "gcc test.c -lfftw3 -lm -o test". When I try to compile the same program with "riscv64-unknown-elf-gcc test.c -lfftw3 -lm -o testpk" it gives me an error 

> test.c:3:10:  fatal error: fftw3.h: No such file or directory  
#include <fftw3.h>

I'm guessing its not seeing the fftw library that I have installed?

Thanks

aswaterman, 2017-06-29:
You need to add ```-I/path``` for the directory containing fftw3.h, and when linking you need ```-L/path``` for the directory containing libfftw3.a.

closed 2017-06-29 by aswaterman

NewPath2015, 2017-06-29:
I have added the path to the command as so "riscv64-unknown-elf-gcc test.c -I/usr/local/include -L/usr/local/lib -lfftw3 -lm -o testpk"

and I'm getting

`/home/myname/Project/rocket-chip/riscv-tools/lib/gcc/riscv64-unknown-elf/7.1.1/../../../../riscv64-unknown-elf/bin/ld: skipping incompatible /usr/local/lib/libfftw3.a when searching for -lfftw3
/home/myname/Project/rocket-chip/riscv-tools/lib/gcc/riscv64-unknown-elf/7.1.1/../../../../riscv64-unknown-elf/bin/ld: cannot find -lfftw3
collect2: error: ld returned 1 exit status`

I did a search but haven't been able to find anything

aswaterman, 2017-06-29:
/usr/local/lib is for host-system  (i.e. x86) libraries - is there really a
file named /usr/local/lib/libfftw3.a that contains RISC-V code?

NewPath2015, 2017-06-30:
I downloaded the fftw 3.3.6 from their website and installed according to the instructions. I guess it installed into the /usr/local/lib and /usr/local/include folders. All I did was just search for the 2 files fftw3.h and libfftw3.a on my system. Sorry I'm pretty new at this so not sure what you meant with libfftw3.a contains RISC-V code. Isn't just part of the fftw library.

aswaterman, 2017-06-30:
.a files contain compiled library code, so they are specific to a given ISA.  If you downloaded and installed them the normal way, they will contain code compatible with your host machine.  If you want to build fftw for RISC-V, you'd need to download the source code and cross-compile it to RISC-V.  I don't know of a good tutorial for that, but as a starting point, try this in the fftw source code directory: ```mkdir build && cd build && ../configure --host= riscv64-unknown-elf && make```

NewPath2015, 2017-06-30:
I compiled the fftw source code with your instructions. Everything is working now, thank you.

aswaterman, 2017-06-30:
np