chipsalliance/rocket-chip #841: Too sparse RAM for Vivado to synthesize data array in DCache

sashimi-yzh, 2017-07-03:
After merging 8aa16a11f31208, Vivado complains with the following messages
```
INFO: [Synth 8-5546] ROM "ram_reg[511]" won't be mapped to RAM because it is too sparse
INFO: [Synth 8-5546] ROM "ram_reg[510]" won't be mapped to RAM because it is too sparse
INFO: [Synth 8-5546] ROM "ram_reg[509]" won't be mapped to RAM because it is too sparse
...
```
And data array in DCache is mapped to about 70,000 LUTs and 145,000 FFs, which makes it impossible to perform implementation in Vivado. I try to implement two rocket cores on zc706, and Vivado tells me that the resource is not enough to perform placement.

This is because the commit above changes [the declaration of `data_arrays`](https://github.com/freechipsproject/rocket-chip/commit/8aa16a11f31208a6db4efa2561a1d7dd60ca4176#diff-93d235ebb1c3e9283d5e758a942a853fR37). This change may be not very friendly to Vivado. [Here](https://forums.xilinx.com/t5/Synthesis/Can-not-mapped-to-RAM-because-it-is-too-sparse/m-p/731899#M19923) is a similar issue on Xilinx forum for reference.

aswaterman, 2017-07-03:
We don't have a solution yet, but the eventual plan is to split the RAMs widthwise as part of backend memory generation.  As a workaround, you can enable ECC (try dataECC = new SECDEDCode, dataECCBytes = 4).  This will change the RAM aspect ratios so Vivado can map it to BRAMs.

michael-etzkorn, 2023-01-12:
Feel free to reopen this issue if this is reproducible. Within chipyard, I can't reproduce this, and rocket-chip doesn't have an internal FPGA flow.

closed 2023-01-12 by michael-etzkorn