chipsalliance/rocket-chip #858: Unable to make DefaultSmallConfig in emulator

NewPath2015, 2017-07-12:
I have built DefaultConfig and DualCoreConfig no problem. Trying to build DefaultSmallConfig with `make -j4 CONFIG=DefaultSmallConfig run` gives me error at this point:

`./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-csr 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-csr.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-mcsr 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-mcsr.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-illegal 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-illegal.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-ma_addr 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-ma_addr.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-ma_fetch 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-ma_fetch.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-sbreak 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-sbreak.out && [ $PIPESTATUS -eq 0 ]
./emulator-freechips.rocketchip.chip-DefaultSmallConfig +max-cycles=100000000 +verbose output/rv64mi-p-scall 3>&1 1>&2 2>&3 | /home/NewPath2015/Project/rocket-chip/riscv-tools/bin/spike-dasm  > output/rv64mi-p-scall.out && [ $PIPESTATUS -eq 0 ]
Makefile:40: recipe for target 'output/rv64mi-p-csr.out' failed
make: *** [output/rv64mi-p-csr.out] Error 1
make: *** Waiting for unfinished jobs....`

I have also tried `make -j4 config=DefaultSmallConfig run` but it ended up just making DefaultConfig version instead.

colinschmidt, 2017-07-14:
Do you have ```$RISCV``` set to your install path for riscv-tools?

NewPath2015, 2017-07-14:
Yes, echo $RISCV outputs `/home/NewPath2015/Project/rocket-chip/riscv-tools`

colinschmidt, 2017-07-14:
Is there anything in ```output/rv64mi-p-csr.out```?

Does the tool version installed at ```$RISCV``` correspond to the submodule pointed to by your rocket-chip?

gthompsonx, 2017-07-28:
I am seeing a similar issue with DefaultSmallConfig:
Makefile:40: recipe for target 'output/rv64mi-p-csr.out' failed
make: *** [output/rv64mi-p-csr.out] Error 13

At the end of output/rv64mi-p-csr.out:
C0:     129288 [0] pc=[000000085c] W[r 0=0991f692fccc6753][0] R[r 0=0000000000000000] R[r18=0000000000000003] inst=[7b200073] DASM(7b200073)
*** FAILED *** (tohost = 13)
C0:     129289 [0] pc=[000000085c] W[r 0=0991f692fccc6753][0] R[r 0=0000000000000000] R[r18=0000000000000003] inst=[7b200073] DASM(7b200073)
*** FAILED *** (code = 13, seed 1501295180) after 129290 cycles

gthompsonx, 2018-04-04:
> @gthompsonx <https://github.com/gthompsonx> can you please close this
> ancient issue?

It says it was closed on Aug 7, 2017.

gthompsonx, 2018-04-04:
Sorry. I was confused. My comment above was for a different issue.

I don't own this issue and didn't submit it.

It's not mine to close.

closed 2018-04-04 by NewPath2015