chipsalliance/rocket-chip #866: Unable to make emulator with new CONFIG

clare7, 2017-07-15:
I couldnt make any new (or even existing CONFIG with modified source code) after I perform make clean.
It will complain missing .fir needed to generate .v file. 

Before tried make clean, I couldnt re-make as it return message "nothing to be done to all" even though I have updated and made changes to my source file.

Any idea?

colinschmidt, 2017-07-17:
Does all your source live in ```src/main/scala```? Thats where the makefile will look updated dependencies so if nothing has changed in there then there will be "nothing to be done".

```make clean``` should not cause any changes like the ones you describe.

clare7, 2017-07-19:
Yes all the source files are located in 
Even after `make clean`, I still have the problem. It seems that the compiler refused to regenerate the `.fir `even though the old ones have been removed after `make clean`.

Somehow I realized the compile did not look into the `src/main/scala`  After several trial and error, I tried to modify the `makefrag` file slightly. From 

```
src_path := src/main/scala
default_submodules := . hardfloat chisel3
chisel_srcs := $(foreach submodule,$(default_submodules) $(ROCKETCHIP_ADDONS),$(shell find $(base_dir)/$(submodule)/$(src_path) -name "*.scala"))
```

to:
```
src_path := src/main/scala
default_submodules := hardfloat chisel3
chisel_srcs := $(foreach submodule,$(default_submodules) $(ROCKETCHIP_ADDONS),$(shell find $(base_dir)/$(submodule)/$(src_path) -name "*.scala"))
```

Basically, I only remove the " . " which is referring to the current directory, i.e. `src/main/scala`

After that, everything works fine again. Even after `make clean`
I dont really understand how this works.

colinschmidt, 2017-07-20:
I'm a bit confused now. You've removed the dependence on ```src/main/scala``` in the makefile and it now rebuilds when you make changes in ```src/main/scala```? This seems backwards to me.

clare7, 2017-07-21:
It will search through the `src_path := src/main/scala` with the removal of `.` from `default_submodules ` as the path is enlisted in` chisel_srcs `

But what I have noticed after several trials is that with `default_submodules := . hardfloat chisel3` , the compiler will search through hardfloat/src/main/scala and `chisel3/src/main/scala`

I am not sure how that happens and I may be wrong as well. Right now removing that  `.` does really solve my problem. Im very confused as well

colinschmidt, 2017-07-21:
Well since you got everything working I'm just going to close this.

closed 2017-07-21 by colinschmidt