chipsalliance/rocket-chip #867: rocket-chip release tag

yuz835, 2017-07-17:
Hi,

The rocket-chip github master branch is under development,
is there any plan to make stable release tags?

xfguo, 2017-08-22:
Agree! That's really important for commercial use.

ianperryman, 2017-09-19:
Yes this would be very helpful for evaluating the RISC-V architecture.  Releases will make it easier to discuss feature content and issues.

jrpetrus, 2018-10-29:
I agree. It's really hard to evaluate the RocketChip master branch. The various submodules do not compiling without considerable hacking. I'm looking at you riscv-tools and chisel3.

bonannojj, 2021-12-07:
A related question (not sure if this justifies a separate issue): Is there any documentation about what versions of the RISC-V spec, and which extensions, the master branch implements?

sequencer, 2022-04-08:
The reason not releasing RC is because RC is so huge to main the API compatibility.
We are considering splitting RC into more small repos #2961 to maintain the API compatibility.
Also there is other needs to clean up dead code and documentation.
We will reach to a concrete plan to releasing RC, but it might not be the current RC, it might be diplomacy, rocket-core, tilelink, AMBA4, and other small components.

closed 2022-04-08 by sequencer