chipsalliance/rocket-chip #877: Unable to make vsim in the latest database

woodyplum, 2017-07-19:
hi, 
     i just check out the latest database and the toolchain from git, and there is no problem when make and run the emulator with both asm-test ans bench test  with the default configuration. but got error when compile the vsim, and i can not find any log file in the vsim, here is the screen log, please help on it, thanks.

rgs,

--------------------------------- error info start ---------------------------------
mkdir -p /home/leewg/projects/rocket-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/leewg/projects/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.fir -o /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.conf
Total FIRRTL Compile Time: 30298.6 ms
cd /home/leewg/projects/rocket-chip/vsim/generated-src && \
	rm -f /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.behav_srams.v && \
	/home/leewg/projects/rocket-chip/vsim/vlsi_mem_gen /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.conf >> /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.behav_srams.v.tmp && \
	mv /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.behav_srams.v.tmp /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.behav_srams.v
cd . && \
	rm -rf csrc && \
	vcs -full64 -notice -line +lint=all,noVCDE,noONGS,noUI -error=PCWM-L -timescale=1ns/10ps -quiet +rad +v2k +vcs+lic+wait +vc+list -CC "-I/local/tools/synopsys/vcs/include" -CC "-I/local/tools/riscv_toolchain/include" -CC "-std=c++11" -CC "-Wl,-rpath,/local/tools/riscv_toolchain/lib" /local/tools/riscv_toolchain/lib/libfesvr.so -sverilog +incdir+/home/leewg/projects/rocket-chip/vsim/generated-src +define+CLOCK_PERIOD=1.0 /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.v /home/leewg/projects/rocket-chip/vsim/generated-src/freechips.rocketchip.chip.DefaultConfig.behav_srams.v /home/leewg/projects/rocket-chip/vsrc/TestDriver.v /home/leewg/projects/rocket-chip/vsrc/SimDTM.v /home/leewg/projects/rocket-chip/vsrc/jtag_vpi.v /home/leewg/projects/rocket-chip/vsrc/plusarg_reader.v /home/leewg/projects/rocket-chip/vsrc/ClockDivider2.v /home/leewg/projects/rocket-chip/vsrc/ClockDivider3.v /home/leewg/projects/rocket-chip/vsrc/AsyncResetReg.v  /home/leewg/projects/rocket-chip/csrc/SimDTM.cc /home/leewg/projects/rocket-chip/csrc/jtag_vpi.c +define+PRINTF_COND=TestDriver.printf_cond +define+STOP_COND=!TestDriver.reset +define+RANDOMIZE_MEM_INIT +define+RANDOMIZE_REG_INIT +define+RANDOMIZE_GARBAGE_ASSIGN +define+RANDOMIZE_INVALID_ASSIGN +libext+.v  +vpi -P /home/leewg/projects/rocket-chip/vsrc/jtag_vpi.tab -CC "-DVCS_VPI" -o ./simv-freechips.rocketchip.chip-DefaultConfig \
	-debug_pp \

/bin/sh: 0: Illegal option -h
make: *** [simv-freechips.rocketchip.chip-DefaultConfig] Error 2

terpstra, 2017-07-19:
Sounds like you /bin/sh points to dash, not bash. Much software, including vcs, does not work with this default.

woodyplum, 2017-07-20:
hi terpstra, 
    thanks for your reply, i just check the shell, it do bash. while when check with vcs command, it response with the same problem, although the dve works fine. i will check it more. 
    any way, thanks for you help.

regards,

woodyplum, 2017-07-20:
This is do the dash problem. Bad news is that after fixed it, the VCS require lower version of GCC, downgrade to 4.4, but not support C++11 which required by rocket, any one has suggestion for the VCS tool version ans gcc versions ?

aswaterman, 2017-07-20:
I believe GCC 4.8 is new enough to compile rocket-chip's dependencies.  Maybe GCC 4.8 is also old enough to work with your version of VCS.  If that fails, I'd recommend upgrading VCS.

woodyplum, 2017-07-21:
got it, thanks for the help.

closed 2017-07-21 by woodyplum