chipsalliance/rocket-chip #879: how to improve branch prediction or to improve compiler？

apolonic, 2017-07-20:
2 traps that I met:
1. M-mode->S-mode->U-mode caused trap_instruction_page_fault
2. memcpy loop repeat like 1024 times caused trap_store/load_page_fault

It is noticed that mcause or scause will be set to 4 and the trap makes the PC->EPC to wait for cache done. That could take hundreds of cycles to wait. 

Could you enlight me about these questions:
Is it a branch prediction design (version) implementation issue? any workaround to improve it? any of the compiler improvement can help?

Thanks,
[test_spike_pk_hello(1).zip](https://github.com/freechipsproject/rocket-chip/files/1161054/test_spike_pk_hello.1.zip)

1. What do you regard trap_store_page_fault(trap_store_access_fault in 6.1.0) occurring on the 1st sd(store dword) instruction of a ld-sd loop that repeated 128 times in pk.memcpy?

Can it be the same rule for trap_load_page_fault occurring on the 1st lbu(load byte unsigned) instruction of a load loop that repeated 14 times in hello.strlen? (trap_load_access_fault in 6.1.0 occurring on the 1st ld instruction in hello.__register_exitproc, no loop in it)

2. trap_instruction_page_fault(trap_instruction_access_fault in 6.1.0) occurred after pk.load_elf which called pk.__do_mmap(map to user vm) and pk.start_user, user space ARE working , but actually not yet done. Is it possible L1(I) cache miss caused the trap? Cause the trap occurred >= 2 times in both versions.
As Rocket software Simulator is simple and in-order processing, can we accept those traps as the inner defects, not the bugs?

aswaterman, 2017-07-20:
This is a software thing, not a hardware thing.  The code and data are demand-paged.  The first time the user program touches a virtual-memory page, it will fault.  (The same sort of thing would happen running a similar program on a computer running Linux instead of PK.)

Just like benchmarking a real piece of hardware, I suggest running for a long time, so these "cold-start" page faults are amortized.

apolonic, 2017-07-20:
Thanks @aswaterman , but do you know how the software would improve it because if the EPC-> PC->EPC keeps returning at the same place, then what else it can execute first to trigger the cache get prepared earlier?

apolonic, 2017-07-20:
oh, I see, the trap program can be modified by the software guy, they can create a flag or timer to watch it and then return the EPC+n to run something else as well as waiting for the cache. Or the system designer can make the compiler to put nops or unreletive instructions to fill that waiting time...

michael-etzkorn, 2023-01-11:
Closed to age. Further questions related to this topic can be directed to StackOverflow or rocket-chip's Github discussion board.

closed 2023-01-11 by michael-etzkorn