chipsalliance/rocket-chip #880: Some comments needed for the diplomacy package

wsong83, 2017-07-20:
I have been trying to understand the diplomacy, TileLink2 and the cake pattern for a couple of months now. Some code segments and definitions have bothered me for quite a while. May be I can have some comments on them, or even better, adding some comments directly into the code would be great.

In diplomacy/Nodes.scala:

- What is the difference between inwards/outwards and upwards/downwards? Sometime I think the inwards and outwards are describing the input and output ports/channels. But I am never certain.

- In the MixedNode class, what are the meanings of oPortMapping, iPortMapping, oStar, iStar, also iKown, oKown. Thesee variables are used to resolve the port mappings and id range assignment I believe. However, due to the use of recursive calling between neighboring nodes, it is really difficult for me to figure out the underlying meanings of these variables.

- The *= and =* assignment is also difficult to understand. Thanks to Wesley's comments, my rough understanding is that, these are used to identify the one to many and many to one connections. However, I cannot easily set up a connection between them and the physical channels. For a code like (I am not sure whether this is legal):
~~~
A :=* B
A :=* C
~~~
can I say this is a (B,C) to A multiplexer? But then there is something like:
~~~
A := B
A := C
~~~

mwachs5, 2017-07-31:
Agree that some comments would be helpful. I'm working on this, you're also welcome to make a PR with your best understanding and we can fill in gaps. 

Here's some answers to your specific Qs:

> What is the difference between inwards/outwards and upwards/downwards? 
Upwards/Downwards refers to the logical flow of parameters. This doesn't necessarily translate to anything in real hardware.

Inwards/Outwards refers to the fact that ever node has two halves, the inwards side towards the Master/Source/Client (what you call it depends on the diplomacy subclass), and the Outward side towards the Slave/Sink/Manager.

The two sets of terms are used clarify the meaning since you can have upwards and downwards parameters on  inwards side of a node, and same for downwards side of a node. 

> In the MixedNode class, what are the meanings of oPortMapping, iPortMapping, 

If a node has several several :=, :*=, and :=* connections, you need to keep track of what nodes actually connect. E.g. for the :=* connection, the same parameters may need to be applied to several edges. This is used for lots of things, like passing parameters and making actual wire connections during the elaboration phase. 

> what are the meaning of oStar, iStar, also iKnown, oKnown

Every time you make a  := connection, you increment your oKnown or iKnown (depending what side you're on). It's an "increment by 1". Every time you make a :=* or :*= connection, you increment your "number of star" connections for the appropriate side. At some point you have to resolve all the stars to know how many connections you have. For example if you do:

```
A :=* B
```
 
It means that B must "know" the value of the star (e.g it knows it has 3 outputs, which it determined elsewhere), A has 1 "iStar", but the value of the iStar is 3 when all is resolved.

If you do 
```
A :*= B
```
It means that A "knows" how many inputs it has and B has to learn what its oStar is worth.

Here's a simple concrete example. (https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/devices/debug/Debug.scala#L400)

```
 intnode :*= dmOuter.intnode
```

This is in the Debug Module, which is written to generate a debug interrupt for each core, without knowing how many cores are in the system. Their debug interrupt sinks just keep connecting to node(s) which eventually gets connected with `intnode` here.  The line gathers all the edges *going into* `intnode` to the single Nexus node in the dmOuter. Nexus nodes don't "know" anything about how many connections they have, they have to "learn" from the other side of the connection, which does know. 

Then in the debug module elaboration phase we can query the size of the edges of the node (https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/devices/debug/Debug.scala#L408)

If we just said (in a multi-core system):

```
intnode := dmOuter.intnode
```

Diplomacy would complain that things don't add up, since lots of cores have connected to `intnode` (which as an IdentityNode has to have the same connections in and out), but there is only one connection coming into it.

Both of your examples above are legal, depending on the type of node that A, B, and C are and whether they support unknown numbers of connections. You can't say that it is a multiplexer, that highly depends on the whole subclass of Diplomacy that you've created and what kinds of nodes A, B, and C are. Those lines of code mean something very different for an IntNode vs a TLNode, and something more different depending on the type of node (Identity? Source? Sink? Nexus?). Suggest looking at the subclasses of Diplomacy, like the protocols in src/main/scala/amba/, to get a more intuitive feeling.

wsong83, 2017-08-01:
Thank you very much Megan! I have misunderstood the meaning of U/D/I/O and the meaning of `:*=` and `:=*`.

For U/D/I/O, I have kind of confused U/D with the Manager/Client relationship. I think one of the confusing places is this TLImp definition in tilelink/Node.scala:

~~~
object TLImp extends NodeImp[TLClientPortParameters, TLManagerPortParameters, TLEdgeOut, TLEdgeIn, TLBundle]
~~~

Here both DI and DO parameters are `TLClientPortParameters` and both UI and UO parameters are `TLManagerPortParameter`. This directly leads me to believe D means client and U means manager. So my misunderstanding starts. I am not sure whether there is a way to make it more clear. With your explanation, I will reread the parts related to see whether I can fully understand them now.

As for `:*=` and `:=*`, I think it would be difficult for people from hardware background to understand for two reasons:

- The diplomacy package, or here the Node/Edge/Bundle system in particular, provides a lazy elaboration process. That is to say, individual modules store their parameters (or interconnect requirements) into the diplomacy Node. Then after the normal `val` has been processed, the `lazy val` enforces an order (which I am not sure) to traverse the interconnect using these logically connect Nodes to resolve actual port connections.
- No matter `:=`, `:*=`, or `:=*`, they all represent one to one or many to many connections. The conceptional data always flow from rhs (output) to lhs (input). Star means, for many to many connections, which side decides the number of connections. Am I right this time? But can a `A := B` connection means many to many as well, IFF A and B have the same number of ports and this assignment to A happens only once?

For the first point, is it possible for SiFive/UCB to provide a simple document describing the lazy process? What I roughly understand now is: for port connections, the assignment statements build up bindings and store the bind callbacks into stacks in each InwardsNode. Then some time later (I am not sure when), these callbacks are called and all parameters for connections are resolved. So finally these connections are used through the bundleIn and bundleOut variable to form the IO in implementation module classes.

Indeed I have been keeping notes through my process of reading the code. I could try to feedback some comments. Will try for a first PR this week.

mwachs5, 2017-08-21:
> Would it be possible to use Node to build a ring?

I believe diplomacy detects loops. 

> Can a node have more than one star type connections on one side (either inner or outer)?

Sure. Imagine somethign like an IntXbar (`A`) which is aggregating interrupts from a variety of sources (`B`). You could certainly imagine something like:

```A :=* B1
A :=* B2
A :=* B3

wsong83, 2017-08-21:
Thank you Megan, I think I have learned enough to start my porting job for lowRISC. Your answers are very helpful.

closed 2017-08-21 by wsong83

jerryhethatday, 2020-11-23:
> > Would it be possible to use Node to build a ring?
> 
> I believe diplomacy detects loops.
> 
> > Can a node have more than one star type connections on one side (either inner or outer)?
> 
> Sure. Imagine something like an IntXBar (`A`) which is aggregating interrupts from a variety of sources (`B#` and `C#`, where B has multiple outputs and C only has one). You could certainly write something like:
> 
> ```
> A :=* B1
> A :=* B2
> A := C1
> A := C2
> ```

is something like A:*= B1    A:*= B2(node A has two BindStar Nodebinds.)  or A1 :=* B A2 =*: B(B has two BindStar connections) legal???
I feel like a node can not have more than one BindStar Nodebinding.  I went thru the InwardNode trait, and there is one field iStar whose comments just indicates"resolved [[BIND_STAR]] binding of inward nodes: how many connections the star represents." if the inwardnode is allowed to have more than 1 Bind_Star, then I feel like the oStar should be a collection? Am I right?@ @wsong83 @mwachs5

jerryhethatday, 2020-11-24:
> I believe diplomacy detects loops.
>
>
> Sure. Imagine something like an IntXBar (`A`) which is aggregating interrupts from a variety of sources (`B#` and `C#`, where B has multiple outputs and C only [...]

Sorry to bother, I wonder if it is this legal to interconnect `A :*= B1  A :=* B2`  I post a question issue in the chipyard repo, Hope you can take a look.https://github.com/ucb-bar/chipyard/issues/725