chipsalliance/rocket-chip #881: Can't have two empty heterogenous bags in a bundle

mwachs5, 2017-07-21:
Not sure if this is a Chisel bug, or something wrong with Heterogenous Bag. 

The following code won't elaborate, it somehow swaps around the heterogenous bags that are empty:

```
class Thing extends Bundle {
  val bla = Bool(INPUT)
  val fa = Bool(OUTPUT)
}
class MyBundleWithEmptyStuff(c: Int) extends Bundle {
  val barNone = HeterogeneousBag(Seq.fill(c) {new Thing()})
  val bazNone = HeterogeneousBag(Seq.fill(c) {new Thing()})
}

class MySimilarBundleWithEmptyStuff(c: Int) extends Bundle {
  val bazNone = HeterogeneousBag(Seq.fill(c) {new Thing()})
  val barNone = HeterogeneousBag(Seq.fill(c) {new Thing()})
}

class InnerModuleWithEmptyThings extends Module {
  val io = new MyBundleWithEmptyStuff(0)

  io.barNone.foreach ( x => printf("Hooray BarNone %d!", x.bla))
  io.bazNone.foreach ( x => printf("Hooray BazNone %d!", x.bla))

  io.bazNone.foreach {x => x.fa := x.bla}
  io.barNone.foreach {x => x.fa := x.bla}
}

class OuterModuleWithEmptyThings extends Module {
  val io = new MySimilarBundleWithEmptyStuff(0)

  val inner = Module(new InnerModuleWithEmptyThings())

  io.barNone <> inner.io.barNone
  io.bazNone <> inner.io.bazNone
 
}
```

This is the FIRRTL that gets created for OuterModuleWIthEmptyThings:

```
 module OuterModuleWithEmptyThings : 
    input clock : Clock
    input reset : UInt<1>
    output io : {barNone : {}}
    
    clock is invalid
    reset is invalid
    io is invalid
    inst inner of InnerModuleWithEmptyThings @[ExampleTop.scala 63:21]
    inner.io is invalid
    inner.clock <= clock
    inner.reset <= reset
    io.barNone <- inner.io.barNone @[ExampleTop.scala 65:14]
    _T_2 <- _T_2 @[ExampleTop.scala 66:14]
```

mwachs5, 2017-07-21:
This test case is on a branch here https://github.com/freechipsproject/rocket-chip/tree/heterogenous_bag_mixup

mwachs5, 2017-07-21:
Wow I really can't spell heterogeneous

terpstra, 2017-07-21:
@jackkoenig My guess is that this problem affects all classes derived from Record that have an empty elements.

mwachs5, 2017-07-21:
If @terpstra is right, I'm working on a test case just using Record to submit to chisel3 repo

mwachs5, 2017-08-15:
Fixed by #956

closed 2017-08-15 by mwachs5