chipsalliance/rocket-chip #883: build C emulator failed!

gchen-viosoft, 2017-07-22:
```grady@qt03:~$ cd work/rocket-chip/emulator/
grady@qt03:~/work/rocket-chip/emulator$ make
mkdir -p /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig
/home/grady/work/rocket-chip/emulator/verilator/install/bin/verilator --cc --exe --top-module TestHarness +define+PRINTF_COND=\$c\(\"verbose\",\"\&\&\"\,\"done_reset\"\) +define+STOP_COND=\$c\(\"done_reset\"\) --assert --output-split 20000 --output-split-cfuncs 20000 -Wno-STMTDLY --x-assign unique -I/home/grady/work/rocket-chip/vsrc -O3 -CFLAGS "-O1 -std=c++11 -I/home/grady/work/rocket-chip/riscv/include -I/home/grady/work/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/grady/work/rocket-chip/csrc/verilator.h" -Mdir /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig \
        -o /home/grady/work/rocket-chip/emulator/emulator-freechips.rocketchip.chip-DefaultConfig /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig.v /home/grady/work/rocket-chip/csrc/emulator.cc /home/grady/work/rocket-chip/csrc/SimDTM.cc -LDFLAGS " -L/home/grady/work/rocket-chip/riscv/lib -Wl,-rpath,/home/grady/work/rocket-chip/riscv/lib -L/home/grady/work/rocket-chip/emulator -lfesvr -lpthread" \
        -CFLAGS "-I/home/grady/work/rocket-chip/emulator/generated-src -include /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig/VTestHarness.h"
%Error: /home/grady/work/rocket-chip/vsrc/plusarg_reader.v:16: syntax error, unexpected IDENTIFIER, expecting STRING
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/local/bin/verilator_bin --cc --exe --top-module TestHarness '+define+PRINTF_COND=$c("verbose","&&","done_reset")' '+define+STOP_COND=$c("done_reset")' --assert --output-split 20000 --output-split-cfuncs 20000 -Wno-STMTDLY --x-assign unique -I/home/grady/work/rocket-chip/vsrc -O3 -CFLAGS '-O1 -std=c++11 -I/home/grady/work/rocket-chip/riscv/include -I/home/grady/work/rocket-chip/csrc -DVERILATOR -DTEST_HARNESS=VTestHarness -include /home/grady/work/rocket-chip/csrc/verilator.h' -Mdir /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig -o /home/grady/work/rocket-chip/emulator/emulator-freechips.rocketchip.chip-DefaultConfig /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig.v /home/grady/work/rocket-chip/csrc/emulator.cc /home/grady/work/rocket-chip/csrc/SimDTM.cc -LDFLAGS ' -L/home/grady/work/rocket-chip/riscv/lib -Wl,-rpath,/home/grady/work/rocket-chip/riscv/lib -L/home/grady/work/rocket-chip/emulator -lfesvr -lpthread' -CFLAGS '-I/home/grady/work/rocket-chip/emulator/generated-src -include /home/grady/work/rocket-chip/emulator/generated-src/freechips.rocketchip.chip.DefaultConfig/VTestHarness.h'
make: *** [emulator-freechips.rocketchip.chip-DefaultConfig] Error 10
grady@qt03:~/work/rocket-chip/emulator$
```

terpstra, 2017-07-22:
You need the newest version of verilator. Probably you have an old one from a previous build. Try building from a clean checkout.

gchen-viosoft, 2017-07-23:
Thank you. It works after updating verilator to 3.904.

closed 2017-07-23 by terpstra