chipsalliance/rocket-chip #884: Error with "make verilog CONFIG=BOOMConfig"

BenjaminLpn, 2017-07-23:
Hi,

I'm getting an error using the cmd-line below (see (1) below)

cd rocket-chip/vsim
make verilog CONFIG=BOOMConfig

Is there something wrong in my cmd-line setting and what's the correction?
I didn't modify any other files like Makefile

Thanks,

Benjamin

(1) screen output from cmd-line above is below:
`mkdir -p /home/lipn/Software/riscv/rocket-chip/vsim/generated-src/
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -cp /home/lipn/Software/riscv/rocket-chip/firrtl/utils/bin/firrtl.jar firrtl.Driver -i /home/lipn/Software/riscv/rocket-chip/vsim/generated-src/rocketchip.BOOMConfig.fir -o /home/lipn/Software/riscv/rocket-chip/vsim/generated-src/rocketchip.BOOMConfig.v -X verilog --infer-rw TestHarness --repl-seq-mem -c:TestHarness:-o:/home/lipn/Software/riscv/rocket-chip/vsim/generated-src/rocketchip.BOOMConfig.conf
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
Exception in thread "main" java.lang.NoSuchMethodError: firrtl.WSubField$.apply(Lfirrtl/ir/Expression;Ljava/lang/String;)Lfirrtl/WSubField;
	at firrtl.connectFields$.apply(Utils.scala:121)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$defaultConnects$1.apply(ReplaceMemMacros.scala:124)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$defaultConnects$1.apply(ReplaceMemMacros.scala:124)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.immutable.List.map(List.scala:285)
	at firrtl.passes.memlib.ReplaceMemMacros.defaultConnects(ReplaceMemMacros.scala:124)
	at firrtl.passes.memlib.ReplaceMemMacros.adaptReader(ReplaceMemMacros.scala:132)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$3.apply(ReplaceMemMacros.scala:109)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$3.apply(ReplaceMemMacros.scala:109)
	at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
	at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
	at scala.collection.Iterator$class.foreach(Iterator.scala:742)
	at scala.collection.AbstractIterator.foreach(Iterator.scala:1194)
	at scala.collection.IterableLike$class.foreach(IterableLike.scala:72)
	at scala.collection.AbstractIterable.foreach(Iterable.scala:54)
	at scala.collection.TraversableLike$class.flatMap(TraversableLike.scala:252)
	at scala.collection.AbstractTraversable.flatMap(Traversable.scala:104)
	at firrtl.passes.memlib.ReplaceMemMacros.createMemModule(ReplaceMemMacros.scala:109)
	at firrtl.passes.memlib.ReplaceMemMacros.updateMemStmts(ReplaceMemMacros.scala:196)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemStmts$3.apply(ReplaceMemMacros.scala:201)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemStmts$3.apply(ReplaceMemMacros.scala:201)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.ir.Block.mapStmt(IR.scala:225)
	at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:16)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:30)
	at firrtl.passes.memlib.ReplaceMemMacros.updateMemStmts(ReplaceMemMacros.scala:201)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemStmts$3.apply(ReplaceMemMacros.scala:201)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemStmts$3.apply(ReplaceMemMacros.scala:201)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.immutable.List.foreach(List.scala:381)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.immutable.List.map(List.scala:285)
	at firrtl.ir.Block.mapStmt(IR.scala:225)
	at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:16)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:30)
	at firrtl.passes.memlib.ReplaceMemMacros.updateMemStmts(ReplaceMemMacros.scala:201)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemStmts$3.apply(ReplaceMemMacros.scala:201)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemStmts$3.apply(ReplaceMemMacros.scala:201)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.ir.Block.mapStmt(IR.scala:225)
	at firrtl.Mappers$StmtMagnet$$anon$1.map(Mappers.scala:16)
	at firrtl.Mappers$StmtMap$.map$extension(Mappers.scala:30)
	at firrtl.passes.memlib.ReplaceMemMacros.updateMemStmts(ReplaceMemMacros.scala:201)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemMods$1.apply(ReplaceMemMacros.scala:207)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$updateMemMods$1.apply(ReplaceMemMacros.scala:207)
	at firrtl.ir.Module.mapStmt(IR.scala:473)
	at firrtl.Mappers$ModuleMagnet$$anon$11.map(Mappers.scala:90)
	at firrtl.Mappers$ModuleMap$.map$extension(Mappers.scala:100)
	at firrtl.passes.memlib.ReplaceMemMacros.updateMemMods(ReplaceMemMacros.scala:207)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$6.apply(ReplaceMemMacros.scala:217)
	at firrtl.passes.memlib.ReplaceMemMacros$$anonfun$6.apply(ReplaceMemMacros.scala:217)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.TraversableLike$$anonfun$map$1.apply(TraversableLike.scala:245)
	at scala.collection.mutable.ResizableArray$class.foreach(ResizableArray.scala:59)
	at scala.collection.mutable.ArrayBuffer.foreach(ArrayBuffer.scala:48)
	at scala.collection.TraversableLike$class.map(TraversableLike.scala:245)
	at scala.collection.AbstractTraversable.map(Traversable.scala:104)
	at firrtl.passes.memlib.ReplaceMemMacros.execute(ReplaceMemMacros.scala:217)
	at firrtl.passes.memlib.ReplSeqMem$$anonfun$8.apply(ReplaceMemTransform.scala:139)
	at firrtl.passes.memlib.ReplSeqMem$$anonfun$8.apply(ReplaceMemTransform.scala:138)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.passes.memlib.ReplSeqMem.run(ReplaceMemTransform.scala:138)
	at firrtl.passes.memlib.ReplSeqMem.execute(ReplaceMemTransform.scala:160)
	at firrtl.Compiler$$anonfun$4$$anonfun$5.apply(Compiler.scala:251)
	at firrtl.Compiler$$anonfun$4$$anonfun$5.apply(Compiler.scala:251)
	at firrtl.Utils$.time(Utils.scala:23)
	at firrtl.Compiler$$anonfun$4.apply(Compiler.scala:251)
	at firrtl.Compiler$$anonfun$4.apply(Compiler.scala:250)
	at scala.collection.LinearSeqOptimized$class.foldLeft(LinearSeqOptimized.scala:124)
	at scala.collection.immutable.List.foldLeft(List.scala:84)
	at firrtl.Compiler$class.compile(Compiler.scala:250)
	at firrtl.VerilogCompiler.compile(LoweringCompilers.scala:136)
	at firrtl.Driver$.execute(Driver.scala:158)
	at firrtl.Driver$.execute(Driver.scala:184)
	at firrtl.Driver$.main(Driver.scala:200)
	at firrtl.Driver.main(Driver.scala)
/home/lipn/Software/riscv/rocket-chip/vsim/Makefrag-verilog:16: recipe for target '/home/lipn/Software/riscv/rocket-chip/vsim/generated-src/rocketchip.BOOMConfig.conf' failed
make: *** [/home/lipn/Software/riscv/rocket-chip/vsim/generated-src/rocketchip.BOOMConfig.conf] Error 1
`

sorear, 2017-07-24:
```
    Blockquotes use lines with three backquotes each,
    as demonstrated here.
    ```

ccelio, 2017-07-24:
Can you please describe the steps to reproduce this error from a clean checkout of the source code?

BenjaminLpn, 2017-07-31:
@ccelio I'm really sorry that I didn't reply to you earlier. I have not reinstalled this environment. If there is any question, I will contact to you. Thanks very much.

BenjaminLpn, 2017-08-03:
@ccelio After reinstall jdk and vcs, this environment works well. Thanks a lot.

closed 2017-08-03 by BenjaminLpn