chipsalliance/rocket-chip #892: Error with "make verilog CONFIG=DefaultRV32Config"

kkamundsen, 2017-07-26:
Hi,

after updating to the newest rocket-chip I got this error while trying to build the verilog. In a clean checkout, I did the following to produce the error:
1. cd vsim
2. make verilog CONFIG=DefaultRV32Config

The (most sensible part of the) stack trace is:
Caused by: java.lang.IllegalArgumentException: requirement failed: Xbar data widths don't match: List(ExampleRocketSystem) has 8B vs List(plic, clint, dmInner, bootrom, error) has 4B

Thanks,

Kai Kristian

colinschmidt, 2017-07-26:
I can reproduce this error so it seems like an issue that probably cropped up in the recent coreplex collapse refactor.

edouard69, 2017-08-01:
Same error here. Has anyone found a solution?

kkamundsen, 2017-08-03:
As a short term solution, I went back to this commit and it works again:
8c92c50d8544d1d6e40b9cbcfaffc271924cdcc4

edouard69, 2017-08-03:
It's working, thanks!

gajinderpanesar, 2017-08-07:
Hello edouard69,
  Are you confiming the 8c92c50 works or the head has been fixed? I guess the latter as I tried the head but I still see the error.
  Thanks

edouard69, 2017-08-07:
Hi, I confirm that the 8c92c50 works. The last version still got the error.

lintonghit, 2017-08-13:
It because bits width of bus was mismatch between memroy(beatBytes = 8=64bits) and bootrom(defined by XLEN=32bits);
You should change MemoryBusParams.beatBytes=4(you can find it in this file: /src/main/scala/coreplex/Configs.scala)while using parameter XLEN=32. Or you should add a TLFragmenter between outside mem and xbar.

gajinderpanesar, 2017-08-17:
Thanks, I will give it a try.

PR #989 (2017-09-15): Clean up various Configs/Fields/Params
Diplomacy now provides the `x"..."` string context, which converts string literals representing hexadecimal numbers into `BigInt`s. You can insert arbitrary underscores, e.g., `x"8000_0000_0000"` instead of `BigInt(0x800000000000)`.

Config now provides optional default values for `Field` objects that are used as lookup keys. However, using the same class names for key objects as the case class name of the value they are holding results in `illegal cyclic reference`, so I'm trying to start walking back that anti-pattern.

Closes #962, closes #892
closed 2017-09-15 by hcook