chipsalliance/rocket-chip #907: Rocket HW FPU support 32bits divSqrt ?

junhongHsu, 2017-07-31:
Hi all,

I modified the configuration as following. 
But got error message when make verilog.
Could anyone tell me if Rocketchip HW FPU don't support 32bits divSqrt or i made the wrong configurations.
Thanks.

class WithRV32 extends Config(
  (pname,site,here) => pname match {
    case XLen => 32

//    case FPUKey => Some(FPUConfig(divSqrt = false))
    case FPUKey => Some(FPUConfig())

    case _ => throw new CDEMatchError
  })

Regards,
Junhong.

sequencer, 2022-04-08:
seems the configuration is wrong, need investigate.

michael-etzkorn, 2022-06-04:
There's just not enough information and the `FPUKey` and `FPUConfig`, they are apart of the `RocketTilesKey`. Closing and junhong can reopen to Stack Overflow if he'd like to follow up on this. 

Keys/Fields and Configs used in RC are part of the Documentation Plan at #2960

closed 2022-06-04 by michael-etzkorn