chipsalliance/rocket-chip #920: UIntToOH1 in PMP.scala

pentin-as, 2017-08-03:
54:
      val lsbMask = mask | ~(((BigInt(1) << lgMaxSize) - 1).U << lgSize)
      val lsbMask = mask | UIntToOH1(lgSize, lgMaxSize)
74:
    !boundMatch(x, ~(((BigInt(1) << lgMaxSize) - 1).U << lgSize)(lgMaxSize-1, 0), lgMaxSize)
    !boundMatch(x, UIntToOH1(lgSize, lgMaxSize), lgMaxSize)
108:
    val lsbMask = ~(((BigInt(1) << lgMaxSize) - 1).U << lgSize)(lgMaxSize-1, 0)
    val lsbMask = UIntToOH1(lgSize, lgMaxSize)

terpstra, 2017-08-03:
What seems to be the problem?

aswaterman, 2017-08-03:
I believe this is a suggestion to write UIntToOH1 instead of (1 << x) - 1.

terpstra, 2017-08-03:
I see. That's a good idea since UIntToOH does not require an adder of any kind after the barrel shifter.

terpstra, 2017-08-03:
Right. Your existing circuit is essentially the same, but I guess a bit less clear.

aswaterman, 2017-08-03:
DC actually picks up the (1<<x)-1 pattern, but the code is cleaner with the change.

commit 5e1b824e (2017-08-03): Use UIntToOH1

Closes #920
PR #921 (2017-08-03): Use UIntToOH1
Closes #920
commit ba4eecc0 (2017-08-03): Use UIntToOH1 (#921)

Closes #920
closed 2017-08-03 by aswaterman