chipsalliance/rocket-chip #945: Rocket AHB port?

amerlo94, 2017-08-09:
Hi all,

my goal is to have a master and a slave AHB port on the Rocket Core. I saw that the core has AHB-Lite support but I do not find any class on traits in order to implement it.

Is there any working configuration for the ports?

Thanks

colin4124, 2017-08-30:
I have worked on this goal  a few months ago. 

1. UnitTest:  https://github.com/colin4124/rocket-chip/blob/rocket-mcu-dev-hasti-wip/src/main/scala/unittest/Configs.scala

2. integrate with Rocket Core:  https://github.com/colin4124/rocket-chip/blob/rocket-mcu-dev-hasti-wip/src/main/scala/mcurocket/RocketTest.scala

I am so sorry for that I have no document, just ugly code. I hope it can help you somehow.

If you want to know more, welcome to give me email. colin4124@gmail.com

12ff7a6, 2022-01-30:
@colin4124 can you please share your /subsystem/Ports.scala? I want to generate AHB memory ports, but I did found any useful pieces of information.

colin4124, 2022-02-03:
@12ff7a6 I'm so sorry, I haven't used rocket since that time and haven't any update for latest version of rocket chip.

sequencer, 2022-04-08:
Need #2953 to illustrate the usage for adding non-tilelink ports.
Closing for now.

closed 2022-04-08 by sequencer