chipsalliance/rocket-chip #966: AXI4 Slave Port

amerlo94, 2017-08-22:
Hi all,

I would like to have a Slave AXI4 port on the rocket in order to access the SystemBus from an AXI4 master device. So I include the trait HasSlaveAXI4Port and HasSlaveAXI4PortBundle in the Top module but in the generated verilog file the slave port has only input pins and no output ones. How could I manage to have a full AXI4 slave port on the Rocket?

Thanks,
Andrea

terpstra, 2017-08-22:
What about HasSlaveAXI4PortModuleImp ? I don't think you need HasSlaveAXI4PortBundle.

amerlo94, 2017-08-22:
I use the following code in my custom Port.scala file:

/** Adds an AXI4 port to the system intended to be a slave on an MMIO device bus */
trait HasSlaveAXI4MMIOPort extends HasSystemBus {
  private val params = p(ExtIn)
  val mmio_slv_axi4 = AXI4BlindInputNode(Seq(AXI4MasterPortParameters(
    masters = Seq(AXI4MasterParameters(
      name = "AXI4 periphery",
      id   = IdRange(0, 1 << params.idBits))))))

  private val fifoBits = 1
  sbus.fromSyncPorts() :=
    TLWidthWidget(params.beatBytes)(
    AXI4ToTL()(
    AXI4UserYanker(Some(1 << (params.sourceBits - fifoBits - 1)))(
    AXI4Fragmenter()(
    AXI4IdIndexer(fifoBits)(
    mmio_slv_axi4)))))
}

/** Common io name and methods for propagating or tying off the port bundle */
trait HasSlaveAXI4MMIOPortBundle {
  implicit val p: Parameters
  val mmio_slv_axi4: HeterogeneousBag[AXI4Bundle]
  def tieOffAXI4SlavePort(dummy: Int = 1) {
    mmio_slv_axi4.foreach { slv_axi4 =>
      slv_axi4.ar.valid := Bool(false)
      slv_axi4.aw.valid := Bool(false)
      slv_axi4.w .valid := Bool(false)
      slv_axi4.r .ready := Bool(true)
      slv_axi4.b .ready := Bool(true)
    }
  }
}

/** Actually generates the corresponding IO in the concrete Module */
trait HasSlaveAXI4MMIOPortModuleImp extends LazyMultiIOModuleImp with HasSlaveAXI4MMIOPortBundle {
  val outer: HasSlaveAXI4MMIOPort
  val mmio_slv_axi4 = IO(outer.mmio_slv_axi4.bundleIn)
}

Same as proposed in the code provided, it is correct? I think that the PortBundle would not make any difference to the pins provided with the port, as it behaves to components connected to the core. Anyway I would make a try.

terpstra, 2017-08-22:
Where did you mix in the traits? You just need to mix the traits into your Top module...

terpstra, 2017-08-22:
Look at the end of RocketCoreplex.scala, where it defines RocketCoreplex and RocketCoreplexModule. Did you mix in the traits in your version of these classes?

terpstra, 2017-08-22:
Sorry. Things have moved around since I last looked in this part of the code. Look in: system/ExampleRocketSystem.scala
You need to add 'with HasSlaveAXI4Port' and 'with HasSlaveAXI4PortModuleImp' into that list, or make your own derived class that has them.

terpstra, 2017-08-22:
Actually, it looks to me like there is already a default master port in the example in rocket-chip/master. Why do you need to change anything at all?

amerlo94, 2017-08-22:
I add both traits in the Top module.

I made that change as I need different names for the components and I wanted to keep the default Rocket system working.

The problem is not that I could not have a Slave port, I got one. The problem is that this port has only input pins. I tried also the default configuration for Rocket, which has an l2_slv_axi4 port, but also that port has only input pins.

terpstra, 2017-08-22:
Ok. The problem is that firrtl is pruning those ports b/c they are not connected in the TestHarness. I'm looking how to solve this using only open source code.

wsong83, 2017-08-22:
This is exactly the reason I had made the wrapper #964 
But I think you might come out an even more compact way of doing it.

terpstra, 2017-08-22:
Try https://github.com/freechipsproject/rocket-chip/pull/968

A proper fix will probably be a week or two out.

amerlo94, 2017-08-22:
Fix #968 did the magic! I will work with this for the moment and thank you for the support.

PR #1027 (2017-10-03): Make use of the new DontTouch annotation
- Const prop cannot shove hartid into the tile
- Keep all dcache resp fields
- Keep all system-level top ports

Closes #964, closes #966
closed 2017-08-23 by amerlo94