chipsalliance/rocket-chip #970: System Global Reset

amerlo94, 2017-08-23:
Hi,

I am still new to the Rocket environment and I am facing some issue on the implementation side.

I have a system where a Rocket core has been instantiated as a Verilog module: the ExampleRocketSystem module has been instantiated and not the upper TestHarness one. As my testbench starts I see that the Rocket is in its reset state due to the fact that the Debug Module global reset is low. As far I understand, I have to write to the Debug Module Control register (0x10) via the DMI interface in order to deassert the reset, but the problem is that the signal dmi_req_ready is low, and this seems to me that the slave (in this case the dmi interface) is busy. 

Thus my question is: how I could deassert the Rocket from its reset state at power-up? And secondly is there any kind of documentation on how the dmi bus works?

Many thanks,
Andrea

mwachs5, 2017-08-23:
Please note that rocket-chip uses "reset", not "reset_n" like a lot of designs. So when "Debug Module global reset is low", it's not trying to reset anything.

To get rocket-chip out of reset, you just need to deassert `reset`. `dmiReset` doesn't play into it, you can keep it asserted forever if you like.

the Debug Module Control register it useful if you want to *assert* reset later, but for "Getting Started" use case, this shouldn't matter. 

If you share a waveform of your clock and reset signals going into ExampleRocketSystem that might help figure out what's going on.

amerlo94, 2017-08-24:
I was not aware of the reset functionality, thanks for the info. Now I am able to read and write in the system bus, but still no via the DMI interface. I attach the waveform, I hope it will help the debugging!

[image]

When i raise req_valid, there is no response on req_ready. It is the right way to think of?

mwachs5, 2017-08-24:
If you want to send something over the DMI, you *do* have to deassert dmiReset.

mwachs5, 2017-08-24:
And tick dmiClock

amerlo94, 2017-08-24:
Ok, right! Now the interface reply as it should, thanks!

mwachs5, 2017-08-25:
dmiClock is a clock. It needs to be ticking (going up and down) in order for any of the other dmi signals to have any meaning. 

Basically the dmi interface is on a completely different clock/reset domain than the rest of rocket-chip.

closed 2017-08-28 by amerlo94