chipsalliance/rocket-chip #978: Run simple code from ROM

amerlo94, 2017-08-29:
Hi,
I am integrating the Rocket core in an FPGA system and I would like to boot the core with some very simple code located in the internal BootROM. So i made the required changes in the bootrom.S file and built the image. Unfortunately the core does not seem to execute that code, but it runs 0x10040, then 0x10044, then jump back to 0x10040 instruction again ant then it seems halted.

Is there any debug module registers that I had to set in order to make the core resume operations? Which kind of boot procedure I have to follow? Sorry but I am quite new to the system.

ffard-lbl, 2017-08-31:
Did you manually modify `bootrom.S`? Did you run the Makefile in `bootrom`? `bootrom.img` does not get updated when you run a new build of `rocket-chip`. You have to make sure to run the bootrom Makefile, then rerun the verilog Makefile to load the bootrom into the Verilog.

amerlo94, 2017-08-31:
I run the Makefile in `bootrom` and see the updates in the `bootrom.img`. I think is more a problem of core reset timing. I saw on the verilog testbench, `TestDriver.v` that the reset is asserted after a certain delay, 777.7. Is there any reason for that? What kind of synchronization do I have to fulfill?

colinschmidt, 2017-08-31:
What is the resetvector set to? Does it point to your rom?

amerlo94, 2017-09-01:
Hi, I solved the issue: it was a problem of register and memory initialization, and in particular for the initialization of the $random functions. Since I instantiate the rocket core in a another environment in respect to the one into the repository, I have to define `RANDOMIZE_REG_INIT` and `RANDOMIZE_MEM_INIT` in order to have a full coverage of initialization values.

closed 2017-09-01 by amerlo94