chipsalliance/rocket-chip #991: Cannot change system bus width

zhemao, 2017-09-12:
For Hwacha, we want to change the width of the system bus from 64 bits to 128 bits. However, when I change `SystemBusParams` to set `beatBytes = 16`, I get the following error message:

    java.lang.reflect.InvocationTargetException
        at sun.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
        at sun.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
        at sun.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
        at java.lang.reflect.Constructor.newInstance(Constructor.java:423)
        at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:50)
        at freechips.rocketchip.util.HasGeneratorUtilities$$anonfun$2.apply(GeneratorUtils.scala:47)
        at chisel3.core.Module$.do_apply(Module.scala:42)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.Driver$$anonfun$elaborate$1.apply(Driver.scala:92)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:240)
        at chisel3.internal.Builder$$anonfun$build$1.apply(Builder.scala:238)
        at scala.util.DynamicVariable.withValue(DynamicVariable.scala:58)
        at chisel3.internal.Builder$.build(Builder.scala:238)
        at chisel3.Driver$.elaborate(Driver.scala:92)
    Caused by: java.lang.IllegalArgumentException: requirement failed: Xbar data widths don't match: List(ExampleRocketSystem) has 8B vs List(plic, clint, dmInner, bootrom) has 16B
        at scala.Predef$.require(Predef.scala:219)
        at freechips.rocketchip.tilelink.TLXbar$$anonfun$12$$anonfun$15.apply(Xbar.scala:55)
        at freechips.rocketchip.tilelink.TLXbar$$anonfun$12$$anonfun$15.apply(Xbar.scala:54)
        at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
        at scala.collection.TraversableLike$$anonfun$flatMap$1.apply(TraversableLike.scala:252)
        at scala.collection.immutable.List.foreach(List.scala:381)
        at scala.collection.TraversableLike$class.flatMap(TraversableLike.scala:252)
        at scala.collection.immutable.List.flatMap(List.scala:344)
        at freechips.rocketchip.tilelink.TLXbar$$anonfun$12.apply(Xbar.scala:54)
        at freechips.rocketchip.tilelink.TLXbar$$anonfun$12.apply(Xbar.scala:48)
        at freechips.rocketchip.diplomacy.MixedNexusNode.mapParamsU(Nodes.scala:376)
        at freechips.rocketchip.diplomacy.MixedNode.iParams$lzycompute(Nodes.scala:230)
        at freechips.rocketchip.diplomacy.MixedNode.iParams(Nodes.scala:229)
        at freechips.rocketchip.diplomacy.MixedNode.edgesIn$lzycompute(Nodes.scala:239)
    java.lang.RuntimeException: Nonzero exit code: 1
        at scala.sys.package$.error(package.scala:27)
        at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
        at sbt.BuildCommon$$anonfun$toError$1.apply(Defaults.scala:2081)
        at scala.Option.foreach(Option.scala:236)
        at sbt.BuildCommon$class.toError(Defaults.scala:2081)
        at sbt.Defaults$.toError(Defaults.scala:39)
        at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:740)
        at sbt.Defaults$$anonfun$runMainTask$1$$anonfun$apply$36$$anonfun$apply$37.apply(Defaults.scala:738)
        at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
        at sbt.$tilde$greater$$anonfun$$u2219$1.apply(TypeFunctions.scala:40)
        at sbt.std.Transform$$anon$4.work(System.scala:63)
        at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
        at sbt.Execute$$anonfun$submit$1$$anonfun$apply$1.apply(Execute.scala:228)
        at sbt.ErrorHandling$.wideConvert(ErrorHandling.scala:17)

I'm a bit confused by this message, since plic, clint, dmInner, and bootrom are hanging off the periphery bus and should still be 8B. Any idea why this is happening?

terpstra, 2017-09-12:
The sbus=>pbus connection has a WidthWidget, so from the point-of-view of the sbus, those devices are now all 16 bytes (128 bit) as you requested. Your problem is the 'ExampleRocketSystem' slave, which is presumably a blind AXI node coming off the system bus. Probably that should be either adjust also to 16B, put through a WidthWidget, or removed.

zhemao, 2017-09-12:
The problem appears to be the memory bus. The width of the memory bus and system bus cannot be mismatched it seems. Could we add a require so that the error message is less confusing?

terpstra, 2017-09-12:
Probably the only reason they have to match is because of the BroadcastHub. Perhaps the default value of BankedL2Params should include a WidthWidget with the appropriate settings.

PR #1015 (2017-09-26): coreplex: clean up coherence manager attachment point
- get `Parameters` from the calling `coreplex`
- allow managers to signal unrecoverable/`halt`
- add a `TLWidthWidget` between system bus and memory bus, behind manager

Closes #991 , closes #1010
closed 2017-09-26 by hcook