chipsalliance/rocket-chip #999: Compile using simulator other than VCS?

ianperryman, 2017-09-19:
I am trying to compile and run the project but do not have access to VCS.  Has anyone tried compiling with any other simulators? 

The issues I have been running into seem related to the version of the vpi_user.h file that is included in csrc/jtag_vpi.c file.

We are trying to compile the csrc files with g++ 

```
11:49:33 [iperryman@eng-05 vsim]$ g++ --version
g++ (Debian 4.9.2-10) 4.9.2
```
getting this error:
```
rocket-chip/csrc/jtag_vpi.c:290:2: error: invalid conversion from ‘void*’ to ‘PLI_INT32 (*)(PLI_BYTE8*) {aka int (*)(char*)}’ [-fpermissive]
```

The vpi_user.h file that is bundled with the simulator has this header indicating that is is from IEEE 1364 2005.  Not sure if VCS is using a more modern version of that file.

Thanks,
Ian

mwachs5, 2017-09-19:
The regressions run using Verilator, not VCS, so it does work with something other than VCS. If you aren't running a config that actually uses the JTAG VPI (no `CONFIG = WithJTAGDTM_...` ), you could remove this from the flow.

I am working on a way to do even the `WithJTAGDTM_` with Verilator, but no ETA on that task completion.

ianperryman, 2017-09-19:
Thank you for the clarification.
Is there any documentation on the RocketChip testing at the top level?  Is a verilog simulator only required for the JTAG?  Perhaps some block diagram illustrating the DUT and the test bench components.   List of tests? 

Thanks,
Ian

sequencer, 2022-04-08:
We do need documentation on the UT.
Which is currently on going at a PLCT intern project.
closing this for now, this can be moved to discussion in the future.

closed 2022-04-08 by sequencer