chipsalliance/rocket-chip #1008: Cannot change CacheBlockBytes to 128

jwang-us, 2017-09-21:
Hi,

Even if using a simple config as following,

```
class CacheTrial extends Config(
  new WithCacheBlockBytes(128) ++
  new DefaultConfig)
```

There will be problems like:

`Caused by: java.lang.IllegalArgumentException: requirement failed: untagBits(13) > pgIdxBits(12)
`
**What is this "pgIdxBits"? It's hard coded as 12. Can we simple increase its size?**

Thanks.

aswaterman, 2017-09-21:
pgIdxBits is an ISA constant for the 4 KiB page size; it cannot be changed.

If you want to increase the cache line size, you will need to decrease the number of sets, so that the size of a set does not exceed 4 KiB.

jwang-us, 2017-09-21:
@aswaterman Thanks.

As I understand, by "the size of a set" you mean in (cache line size) x (# of sets). Then the # of ways limits how many virtual pages we can hold. But why do we need to have this limitation? Could you please help me to understand this a little bit more? Thank you.

closed 2017-09-21 by jwang-us

terpstra, 2017-09-21:
We use a virtually-indexed physically-tagged (VIPT) L1 cache. In this configuration, if your sets*block size is larger than a virtual memory page, you can end up storing the same physical address at two different positions in the cache, a "synonym". Besides wasting L1 space, this can cause trouble if you update only one of the copies.

@aswaterman it occurs to me that we could have 8kB sets if the L1 presented itself as two logical masters (an odd and even master). The L2 would keep two distinct directory bit for the L1 and shoot-down the synonym for you automatically.

aswaterman, 2017-09-21:
@terpstra yeah, that's equivalent to the MIPS R10K design I mentioned the other day, but your solution avoids the nasty abstraction violation.  Nice.