chipsalliance/rocket-chip #1011: Tiny Config benchmark problem

esolera, 2017-09-21:
I tried to run the benchmarks in the Tiny Config with the next instruction: 
make CONFIG=TinyConfig -j1 run-bmark-tests

And I got the next error:
mkdir -p ./output
ln -fs /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/riscv64-unknown-elf/share/riscv-tests/benchmarks/median.riscv output/median.riscv
./emulator-freechips.rocketchip.system-TinyConfig +max-cycles=100000000 +verbose output/median.riscv 3>&1 1>&2 2>&3 | /home/edgar/Desktop/RISCV_FOLDERS/rocket-chip/riscv-tools/bin/spike-dasm  > output/median.riscv.out && [ $PIPESTATUS -eq 0 ]
Makefile:40: recipe for target 'output/median.riscv.out' failed
make: *** [output/median.riscv.out] Error 1

When I checked the median.riscv.out I saw this fail : 
C0:     544204 [0] pc=[0000085c] W[r 0=00000000][0] R[r 0=00000000] R[r18=00000003] inst=[7b200073] dret (args unknown)
*** FAILED *** (tohost = 1337)
C0:     544205 [0] pc=[0000085c] W[r 0=00000000][0] R[r 0=00000000] R[r18=00000003] inst=[7b200073] dret (args unknown)
*** FAILED *** (code = 1337, seed 1506022825) after 544206 cycles

How Can I run becnhmarks in the 32 bit configuration?
PD: All benchmarks compiled with riscv32-unknown-elf-gcc intruction

aswaterman, 2017-09-21:
Can you paste the entire log into a gist or upload it somewhere?

esolera, 2017-09-22:
I share you a the files in dropbox,  use this link to access them:
https://www.dropbox.com/sh/cvqezzxmr9x3dq9/AACohIhNah3tRQqWZZiUPVjpa?dl=0

files in the dropbox:
median.riscv.out 
median.riscv.dump : from /rocket-chip/riscv-tools/riscv64-unknown-elf/share/riscv-tests/benchmarks
build.log : from /rocket-chip/riscv-tools/riscv-tests/build (this is the build when I use make to compile benchmarks from 32 bits and this is the benchmarks I tried to run in tinyconfig)

thanks for help me. If you need something else, only say me.

aswaterman, 2017-09-22:
TinyConfig only has a 16 KiB scratchpad, and these benchmarks require substantially more memory (like 256 KiB).  I think if you increase the size of the scratchpad in the configuration by 16x it will pass see (https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/coreplex/Configs.scala#L85), but I have no idea if that will cause other problems!

esolera, 2017-09-23:
Thanks for the help. I increase scratchpad and work well.

closed 2017-09-26 by hcook