chipsalliance/rocket-chip #1013: rv64ui-p-sd : The simulation success , but .data contents didn't change.

kelayamatoz, 2017-09-22:
I have done following test:
  cd vsim
  make output/rv64ui-p-sw.vpd
  make output/rv64ui-p-sd.vpd

when checking the rv64ui-p-sw.vpd, the .data contents changed as expected.
but when checking the rv64ui-p-sd.vpd, the .data contents didn't change.

aswaterman, 2017-09-22:
If you think you've observed a bug, can you propose a test that exposes it? It sounds like you're misunderstanding the micro architecture.

kelayamatoz, 2017-09-22:
[image]
[image]

aswaterman, 2017-09-22:
I stand by my original question

kelayamatoz, 2017-09-22:
Yes, I just start to understanding the micro architecture. I am thinking either of them should have similar behavior. 
sorry for my poor english. for your first question, I know every word but can't understand it. Do you want detail steps to show the problem?

mwachs5, 2017-09-22:
@kelayamatoz , I think what @aswaterman is asking is:

1) What exact test did you run? What configuration and what test? (I think you answered that above)

2) What did you expect to happen?

3) What actually happened, and how is that different from what you expected to happen?

In your original question, you said "the .data contents changed as expected". Not sure exactly what you mean "as expected", can you be more clear?

closed 2020-02-09 by mwachs5