chipsalliance/rocket-chip #1016: Core Debug Mode

amerlo94, 2017-09-26:
Hi all,
as stated in the Debug Spec v0.13 each implementation of a RISC-V core could have a different way to enter in Debug Mode. So I would kindly ask you in which way I can enter debug mode for the Rocket core?

Thanks

mwachs5, 2017-09-26:
As in all implementations that follow the spec, you can enter debug mode by writing to DMCONTROL.HALTREQ bit in the Debug Module. In rocket-chip, this raises a special debug interrupt to the  Core, which then jumps to the Debug Module ROM and begins executing. In fact, the appendix A.2 of the Debug Spec v13 is a pretty good description of how it works in rocket-chip.

Here's where you can see the debug trap vector getting set:

https://github.com/freechipsproject/rocket-chip/blob/master/src/main/scala/rocket/CSR.scala#L494

amerlo94, 2017-09-26:
Ok.

The problem was that I was trying to access CSR registers from the Debug Module, as I missed the line in the A.2 appendix of the spec debug. I entered correctly the debug mode for the core, but I was trying to do a not permitted operation. As written in the appendix, I will enter instructions in the program buffer in order to read CSR registers.

 So thank you so much for the spot.

closed 2017-09-26 by amerlo94