chipsalliance/rocket-chip #1040: Some AHB signals missing in generated Verilog

adich0, 2017-10-10:
Hello,

I'm attempting to swap out the default AXI interface with an AHB interface. At one point, I was able to generate Verilog with an AHB interface, using a version of the repository dating back to July 28, 2017 (commit  573890e102471bee1e0b60e0dec48fa5c2a7e853) and tweaking Ports.scala. However, I recently updated my sandbox to commit d6766a8c68529fd1511fc93f27e327ea773e2939, made similar tweaks to Ports.scala, and noticed that the generated Verilog has a few AHB signals missing, namely hready, hreadyout, hprot, and hburst. 

I've been digging around for a few days to try to figure out what might be causing this, but have not found anything that's immediately obvious. Could someone point me in the right direction? 

Thanks

terpstra, 2017-10-10:
It's probably pruning them because they are not used in the TestHarness. @jackkoenig We need to fix the generated output so that top-level signals don't get optimized away. There was an attempt to fix this, but it didn't work out.

Try using this: https://github.com/freechipsproject/rocket-chip/pull/968

adich0, 2017-10-10:
Thanks @terpstra, that workaround did generate the missing AHB IO at the DUT. 

I sort of understand why it might prune out some of the output signals (probably because they're implemented as constants?), but why would the input hreadyout get removed? In the older version of the repository that I was using, I did see that there was a corresponding wire connecting the signal to the TLToAHB block, but this seems to no longer be the case.

mwachs5, 2017-10-10:
If they were not driven in the TestHarness, then Chisel elaboration probably tied them to 1 or 0, then they got constant-propagated inside the design.

terpstra, 2017-10-11:
Could you also try https://github.com/freechipsproject/rocket-chip/pull/1033, please? If that fixes this issue, I'd like to merge it.

closed 2017-10-11 by terpstra

adich0, 2017-10-24:
I think there might be some issue with the hready/hreadyout signals on the master side of things. If I understand diagram 4-2 of the AMBA3 AHB-Lite specification correctly, there is a single hready signal that goes to the master. However, the generated Verilog shows an hreadyout signal as an input to the master, and an hready as an output from the master.

terpstra, 2017-10-25:
The rocket AHB bridge is a master + address decoder in one. Therefore, the signals are correct.

PR #1033 (2017-10-11): Use chisel3.experimental.dontTouch (take 2)
Don't remove top level ports from rocket. Closes #1040.