chipsalliance/rocket-chip #1041: I tried to run the emulator in Centos but fail

esolera, 2017-10-10:
When i run : 
make  -j1 run-bmark-tests

but fail with the next message:

[info] Loading project definition from /mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/firrtl/project
usage: git ls-remote [--heads] [--tags]  [-u <exec> | --upload-pack <exec>] <repository> <refs>...
java.util.NoSuchElementException: head of empty stream
        at scala.collection.immutable.Stream$Empty$.head(Stream.scala:1028)
        at scala.collection.immutable.Stream$Empty$.head(Stream.scala:1026)
        at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$11.apply(SbtGit.scala:138)
        at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$11.apply(SbtGit.scala:122)
        at sbt.Init$Value$$anonfun$apply$13.apply(Settings.scala:609)
        at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$constant$1.apply(INode.scala:163)
        at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$constant$1.apply(INode.scala:163)
        at sbt.EvaluateSettings$MixedNode.evaluate0(INode.scala:175)
        at sbt.EvaluateSettings$INode.evaluate(INode.scala:135)
        at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$submitEvaluate$1.apply$mcV$sp(INode.scala:69)
        at sbt.EvaluateSettings.sbt$EvaluateSettings$$run0(INode.scala:78)
        at sbt.EvaluateSettings$$anon$3.run(INode.scala:74)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
[error] java.util.NoSuchElementException: head of empty stream
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore?

aswaterman, 2017-10-10:
Is centos packaged with a really old version of git?  It looks like a failure within sbt when trying to download a git package.

terpstra, 2017-10-10:
sbt is really flaky. I've had similar looking problems when I accidentally ran two builds in the same directory at the same time. Then it got stuck and would never compile again. Try blowing away all generated files in your repository.

esolera, 2017-10-11:
I tried to reinstall the tool but same problem. I tried to upgrade git but say have the last version. 

make -C /mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/sbt-launch.jar" root_dir=/mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/firrtl build-scala
make[1]: Entering directory `/mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/sbt-launch.jar "assembly"
OpenJDK 64-Bit Server VM warning: ignoring option MaxPermSize=256M; support was removed in 8.0
[info] Loading project definition from /mnt/vol_NFS_Zener/WD_ESPEC/RISCV/rocket-chip/firrtl/project
usage: git ls-remote [--heads] [--tags]  [-u <exec> | --upload-pack <exec>] <repository> <refs>...
java.util.NoSuchElementException: head of empty stream
        at scala.collection.immutable.Stream$Empty$.head(Stream.scala:1028)
        at scala.collection.immutable.Stream$Empty$.head(Stream.scala:1026)
        at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$11.apply(SbtGit.scala:138)
        at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$11.apply(SbtGit.scala:122)
        at sbt.Init$Value$$anonfun$apply$13.apply(Settings.scala:609)
        at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$constant$1.apply(INode.scala:163)
        at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$constant$1.apply(INode.scala:163)
        at sbt.EvaluateSettings$MixedNode.evaluate0(INode.scala:175)
        at sbt.EvaluateSettings$INode.evaluate(INode.scala:135)
        at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$submitEvaluate$1.apply$mcV$sp(INode.scala:69)
        at sbt.EvaluateSettings.sbt$EvaluateSettings$$run0(INode.scala:78)
        at sbt.EvaluateSettings$$anon$3.run(INode.scala:74)
        at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
        at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
        at java.lang.Thread.run(Thread.java:748)
[error] java.util.NoSuchElementException: head of empty stream
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore?

I think its a problem with sbt but I dont know what i can do.

seldridge, 2017-10-19:
I'm pretty sure this is just your version of `git`. I'm on RHEL 6.9 which should be similar to your environment. I have to maintain locally installed versions of specific tools to get things to work. One of these tools is `git`. `git ls-remote` requires arguments for earlier versions, but will happily run without them in newer versions. See:
```
seldridge@laptop:[master]~/repos/github.com/freechipsproject/rocket-chip$
> /usr/bin/git --version && /usr/bin/git ls-remote; git --version && git ls-remote | head -n3
git version 1.7.1
usage: git ls-remote [--heads] [--tags]  [-u <exec> | --upload-pack <exec>] <repository> <refs>...
git version 2.12.0
From git@github.com:freechipsproject/rocket-chip
3b36dda9e1102cc4b1160f6ad6431ca4011ef661	HEAD
39023bed047535db7336aafd00393991054532f6	refs/heads/RVFI
7006e02885f8c7a4b8e2cbe0c198fcfefc07d9ea	refs/heads/Scala212-almostcompatible
```

Summarily, RHEL/CentOS are a pain to do development on because of old packages. Sometimes you're stuck, though. There are a lot of ways you can handle this:
  * If you have root, you can download the tool (possibly build it from source) and install in the root directory (not the best option unless you're the only one on the system)
  * Install in your home directory and then prepend that directory to your `$PATH` (note: my personal solution is similar to [this using `stow` for local "package management"](http://blog.felixriedel.com/2012/05/installing-packages-into-home-with-stow/))
  * Use the Linux version of `homebrew`, [`linuxbrew`](http://linuxbrew.sh/), to get something like home directory package management
  * Find a repo that has more up-to-date versions of pacakges for CentOS and use that (maybe EPEL)
  * Complain to your sysadmins that old packages are making your life hell
  * Don't use CentOS

So that this is coalesced, I think this is the same issue as:
  * https://github.com/freechipsproject/firrtl/issues/677
  * https://github.com/freechipsproject/chisel3/issues/707.

esolera, 2017-10-23:
you are correct, thanks, The problem is solve.

closed 2017-10-23 by esolera