chipsalliance/rocket-chip #1057: assertion fails in verilator

wsong83, 2017-10-18:
This is probably a long short.

So I am not talking about this rocket-chip. I have recently ported a relatively new rocket-chip to the lowRISC setting. I can run through the riscv-tests using VCS.

However, when I try to do the same using Verilator (I am using the current master branch from veripool), some assertions fails.
Even more strange, if I turn on the VCD generation in Verilator, the failed test case can pass.

Right now I am not really sure whether this is a Verilator bug or the assertion catches some race conditions or (may be more likely) something in my port is wrong.

May I know that, does the currently rocket-chip run through Verilator regularly?

mwachs5, 2017-10-18:
The current rocket-chip runs through verilator on every Pull Request, using
the Travis Regressions: https://travis-ci.org/freechipsproject/rocket-chip

Note that the rocket-chip regressions build verilator as part of the
regression, so maybe you should make sure you are using the same Verilator
version?

wsong83, 2017-10-18:
Thanks Megan! That confirms I must have some hiccups in my port. I will dig it up then.

closed 2017-10-18 by wsong83