chipsalliance/rocket-chip #1058: cannot build new macros dependency separately

edcote, 2017-10-18:
This is a question and/or bug for @terpstra  wrt to your recent addition of file `macros/src/main/scala/ValName.scala` and it's inclusion in `project/build.scala`:

~~~~
lazy val chisel = project in file("chisel3")
lazy val hardfloat  = project.dependsOn(chisel)
lazy val macros = project in file("macros") // <-- new
lazy val rocketchip = (project in file(".")).settings(chipSettings).dependsOn(chisel, hardfloat, macros)
~~~~

Unlike chisel and hardfloat, I am unable (or do not know how), to build a separate jar file for this dependency.

The result is a build error, downstream, in my code:
`[error] (*:update) sbt.ResolveException: unresolved dependency: berkeley#macros_2.11;1.2: not found`

Here is output of `sbt depedencyTree` showing ability to reference the other dependencies.

~~~~
[info] Done updating.
[info] com.pnlsys:rocketparts_2.11:0.1 [S] // my code
[info]   +-berkeley:rocketchip_2.11:1.2 [S]
[info]   | +-berkeley:macros_2.11:1.2 (error: not found)
[info]   | +-edu.berkeley.cs:chisel3_2.11:3.1-SNAPSHOT [S]
[info]   | | +-com.github.scopt:scopt_2.11:3.6.0 [S]
[info]   | | +-edu.berkeley.cs:firrtl_2.11:1.1-SNAPSHOT [S]
[..]
~~~~

How to build a jar for macros_2.11 separately?

edcote, 2017-10-18:
So, here's a workaround.  Create new file `macros/build.sbt` with the following content:

~~~~
name := "macros"

version := "1.2"

organization := "berkeley"

scalaVersion := "2.11.11"

libraryDependencies ++= Seq("org.scala-lang" % "scala-reflect" % scalaVersion.value)
~~~~

There's some legacy stuff going on in the project's build flow, so I don't feel compelled to create a pull request and add this file.

We can discuss here, then take action as necessary.

edcote, 2017-10-18:
In case you are interested, here is my use case :

~~~~
cd $ROCKETCHIP
git pull
git submodule update --init --recursive

cd $ROCKETCHIP/firrtl
sbt publishLocal

cd $ROCKETCHIP/chisel3
sbt publishLocal

cd $ROCKETCHIP/hardfloat
sbt -DchiselVersion="latest.release" publishLocal

# new step
cd $ROCKETCHIP/macros
sbt publishLocal

cd $ROCKETCHIP
sbt publishLocal
~~~~

edcote, 2018-03-15:
<ping>

Thoughts on adding build.sbt under macros?

Alternatively, know of any users that are importing rocketchip project into another project, without using publishLocal?

terpstra, 2018-03-21:
I don't know what the problem is or how to fix it ...

edcote, 2018-03-21:
its (somewhat) related to part of the discussion in https://github.com/freechipsproject/rocket-chip/pull/1230 @jackkoenig 

dug some more, my original issue was addressed in 
https://github.com/freechipsproject/rocket-chip/commit/b914564a62d73ff2abe433fbc782639dae5f346e#diff-fdc3abdfd754eeb24090dbd90aeec2ce

I am closing

closed 2018-03-21 by edcote