chipsalliance/rocket-chip #1061: Fail run the test with vsim

woodyplum, 2017-10-19:
After update the database few days before, the test with VCS always fail in vsim directory, please help what should i try for the simulation: 
platfort : 14.04.1-Ubuntu, with gcc 4.4 and gcc 4.8.4
change the default GCC to 4.8.4 for the C++11 requirement. 
after compile and install the riscv-tools without errors, the test in emulator always can pass, while the vsim directory always fail. 

**when make in the vsim directory with system GCC 4.8, the error infor is** : 
g++ -w  -pipe -fPIC -I/local/tools/synopsys/vcs-mx_vL-2016.06/include -I/local/tools/riscv_test/include -std=c++11 -Wl,-rpath,/local/tools/riscv_test/lib -DVCS_VPI -O -I/local/tools/synopsys/vcs-mx_vL-2016.06/include    -c /home/admin/xfer/temp/rocket-chip/csrc/SimDTM.cc
gcc -w  -pipe -fPIC -I/local/tools/synopsys/vcs-mx_vL-2016.06/include -I/local/tools/riscv_test/include -std=c++11 -Wl,-rpath,/local/tools/riscv_test/lib -DVCS_VPI -O -I/local/tools/synopsys/vcs-mx_vL-2016.06/include    -c /home/admin/xfer/temp/rocket-chip/csrc/jtag_vpi.c
cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C [enabled by default]
cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C [enabled by default]
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckCallocFromGroupFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsSetSmallCaching'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `initNativeUcli'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsGroupSize'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `storeMiscTfCb'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsHighMemory'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsStrdupFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `setIclCmd_ResVars'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `shutDownNativeUcli'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCallocFromGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `isUcliEngineStarted'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckStrdupFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsFreeFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedPopGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsGroupGetActive'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `registerOutOfMemHandle'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsFreeGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedPushGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckFreeFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsSetHook'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `printIclBeginMarker'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `setEnvPair'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckReallocFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `storeSimLogFileHandle'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `waitForUserInput'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `execMhpiTclCommand'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `miHeapProf_init'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckCallocFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snps__StrdupFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `registerPreRestartFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsMallocFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsMallocFromGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsPrintf'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsSetOutOfMemoryRoutine'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `executeTclProc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsMemSetLogFileHandle'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `miHeapOn'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckMallocFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `printIclEndMarker'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `vcsRunUcliErrorScript'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsDoStackUnwind'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `execMhpiTclCommandNoPrint'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCallocFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsucli.so: undefined reference to `breakNativeTclLoop'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsMemManagerIsActive'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsGetHook'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCheckMallocFromGroupFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsReallocFunc'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsReallocFromGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsGetMemBytes'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsNamedAllocateGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsLowMemory'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsCurrentGroup'
/local/tools/synopsys/vcs-mx_vL-2016.06/linux64/lib/libvcsnew.so: undefined reference to `snpsOutOfMem'
collect2: error: ld returned 1 exit status
make[1]: *** [product_timestamp] Error 1
make[1]: Leaving directory `/home/admin/xfer/temp/rocket-chip/vsim/csrc'
Make exited with status 2
make: *** [simv-freechips.rocketchip.system-DefaultConfig] Error 2

[... 17 lines cut ...]
ln -fs /local/tools/riscv_test/riscv64-unknown-elf/share/riscv-tests/isa/rv64ui-v-subw output/rv64ui-v-subw
cd . && ./simv-freechips.rocketchip.system-DefaultConfig -q +ntb_random_seed_automatic +verbose +max-cycles=100000000 output/rv64um-v-mul 3>&1 1>&2 2>&3 | /local/tools/riscv_test/bin/spike-dasm  > output/rv64um-v-mul.out && [ $PIPESTATUS -eq 0 ]
cd . && ./simv-freechips.rocketchip.system-DefaultConfig -q +ntb_random_seed_automatic +verbose +max-cycles=100000000 output/rv64um-v-mulh 3>&1 1>&2 2>&3 | /local/tools/riscv_test/bin/spike-dasm  > output/rv64um-v-mulh.out && [ $PIPESTATUS -eq 0 ]

Note-[RT_BS] Larger stack needed
  Note: Bumping stack limit from 8192 to 8193 Kbytes.

Note-[RT_BS] Larger stack needed
  Note: Bumping stack limit from 8192 to 8193 Kbytes.

Command line: ./simv-freechips.rocketchip.system-DefaultConfig -q +ntb_random_seed_automatic +verbose +max-cycles=100000000 output/rv64um-v-mul

--- Stack trace follows:

Dumping VCS Annotated Stack:
No context available
make: *** [output/rv64um-v-mul.out] Error 1
make: *** Waiting for unfinished jobs....
Command line: ./simv-freechips.rocketchip.system-DefaultConfig -q +ntb_random_seed_automatic +verbose +max-cycles=100000000 output/rv64um-v-mulh

--- Stack trace follows:

Dumping VCS Annotated Stack:
No context available
make: *** [output/rv64um-v-mulh.out] Error 1

zhangyl4991, 2017-10-28:
Did you solve the problem? I have the same problem now.

colinschmidt, 2017-10-28:
Did you delete the intermediate files after the failed compile? Perhaps you could try a different version of vcs? The empty stack trace isn't going to be easy to debug.

bhyou, 2017-11-09:
@zhangyl4991 you can try make  run CONFIG=DefaultFPGAConfig

hoangt, 2018-01-17:
I used VCS-2013 and run "make run" in rocket-chip/vsim
1) Verilog is generated correctly
2) Enter simulation phase, error messages are show

> rm -f .././simv-freechips.rocketchip.system-DefaultConfig.daidir/libXmap.so
> ld -shared -rpath='$ORIGIN' -rpath=././simv-freechips.rocketchip.system-DefaultConfig.daidir  -o .././simv-freechips.rocketchip.system-DefaultConfig.daidir/libXmap.so xmap0.so
> cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C [enabled by default]
> 5NrIB_d.o:(.data+0x418): undefined reference to `send_result_to_server'
> 5NrIB_d.o:(.data+0x460): undefined reference to `check_for_command'
> collect2: error: ld returned 1 exit status
> make[1]: *** [product_timestamp] Error 1
> make[1]: Leaving directory `/home/hoangt/WORK/TOOLS/risc-v/rocket-chip/vsim/csrc'

closed 2018-02-26 by hcook

SunYongshuai, 2018-03-21:
Did you solve the problem? I have the same problem now.
And if I use VCS alone, I can use -cc=gcc4.4 -cpp g++-4.4 to solve the problem. But what should I do in the    Rocket-chip Project?

SunYongshuai, 2018-03-21:
And could you share your vcs_gnu package to me at sunyongshuai@outlook.com. Thanks a lot.

SunYongshuai, 2018-03-21:
Or reference:
https://github.com/freechipsproject/rocket-chip/issues/21
add "-LDFLAGS -Wl,--no-as-needed" to the vcs invocation at Makefile

ujasthakkar93, 2019-05-15:
> I used VCS-2013, GCC-4.8.3 and run "make run" in rocket-chip/vsim
> 
>     1. Verilog is generated correctly
> 
>     2. Enter simulation phase, error messages are shown as below. Am I using an incompatible version of VCS or other issues?
> 
> 
> > rm -f .././simv-freechips.rocketchip.system-DefaultConfig.daidir/libXmap.so
> > ld -shared -rpath='$ORIGIN' -rpath=././simv-freechips.rocketchip.system-DefaultConfig.daidir  -o .././simv-freechips.rocketchip.system-DefaultConfig.daidir/libXmap.so xmap0.so
> > cc1: warning: command line option '-std=c++11' is valid for C++/ObjC++ but not for C [enabled by default]
> > **> 5NrIB_d.o:(.data+0x418): undefined reference to `send_result_to_server' 5NrIB_d.o:(.data+0x460): undefined reference to `check_for_command'**
> > collect2: error: ld returned 1 exit status
> > make[1]: *** [product_timestamp] Error 1
> > make[1]: Leaving directory `/home/hoangt/WORK/TOOLS/risc-v/rocket-chip/vsim/csrc'

Hi guys,
if you are facing this problem again and again tha check your file or folder name there will not be a space in that name. if there is a space than remove it and than recompile. thnks