chipsalliance/rocket-chip #1064: Verilog build failing

vinod1993, 2017-10-23:
I tried building rocket to generate the verilog sources under vsim/ directory, but I am getting the following error
make -C /home/vinod/repos/freechips/rocket-chip/firrtl SBT="java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/vinod/repos/freechips/rocket-chip/sbt-launch.jar" root_dir=/home/vinod/repos/freechips/rocket-chip/firrtl build-scala
make[1]: Entering directory `/home/vinod/repos/freechips/rocket-chip/firrtl'
java -Xmx2G -Xss8M -XX:MaxPermSize=256M -jar /home/vinod/repos/freechips/rocket-chip/sbt-launch.jar "assembly"
[info] Loading project definition from /home/vinod/repos/freechips/rocket-chip/firrtl/project
SLF4J: Failed to load class "org.slf4j.impl.StaticLoggerBinder".
SLF4J: Defaulting to no-operation (NOP) logger implementation
SLF4J: See http://www.slf4j.org/codes.html#StaticLoggerBinder for further details.
java.security.AccessControlException: access denied ("java.io.FilePermission" "/home/vinod/repos/freechips/rocket-chip/firrtl/src/test/resources/integration" "readlink")
	at java.security.AccessControlContext.checkPermission(AccessControlContext.java:472)
	at java.security.AccessController.checkPermission(AccessController.java:685)
	at sun.nio.fs.UnixFileSystemProvider.readSymbolicLink(UnixFileSystemProvider.java:487)
	at java.nio.file.Files.readSymbolicLink(Files.java:1384)
	at org.eclipse.jgit.util.FileUtils.readSymLink(FileUtils.java:435)
	at org.eclipse.jgit.util.FS.readSymLink(FS.java:695)
	at org.eclipse.jgit.treewalk.FileTreeIterator$FileEntry.openInputStream(FileTreeIterator.java:392)
	at org.eclipse.jgit.treewalk.WorkingTreeIterator.idBufferBlob(WorkingTreeIterator.java:360)
	at org.eclipse.jgit.treewalk.WorkingTreeIterator.idBuffer(WorkingTreeIterator.java:287)
	at org.eclipse.jgit.treewalk.AbstractTreeIterator.getEntryObjectId(AbstractTreeIterator.java:437)
	at org.eclipse.jgit.treewalk.WorkingTreeIterator.contentCheck(WorkingTreeIterator.java:993)
	at org.eclipse.jgit.treewalk.WorkingTreeIterator.isModified(WorkingTreeIterator.java:923)
	at org.eclipse.jgit.treewalk.filter.IndexDiffFilter.include(IndexDiffFilter.java:227)
	at org.eclipse.jgit.treewalk.filter.AndTreeFilter$Binary.include(AndTreeFilter.java:131)
	at org.eclipse.jgit.treewalk.TreeWalk.next(TreeWalk.java:822)
	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:437)
	at org.eclipse.jgit.lib.IndexDiff.diff(IndexDiff.java:377)
	at org.eclipse.jgit.api.StatusCommand.call(StatusCommand.java:148)
	at com.typesafe.sbt.git.JGit.hasUncommittedChanges(JGit.scala:88)
	at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$10$$anonfun$apply$10.apply(SbtGit.scala:121)
	at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$10$$anonfun$apply$10.apply(SbtGit.scala:121)
	at com.typesafe.sbt.git.DefaultReadableGit.withGit(ReadableGit.scala:39)
	at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$10.apply(SbtGit.scala:121)
	at com.typesafe.sbt.SbtGit$$anonfun$buildSettings$10.apply(SbtGit.scala:121)
	at scala.Function1$$anonfun$compose$1.apply(Function1.scala:47)
	at sbt.EvaluateSettings$MixedNode.evaluate0(INode.scala:175)
	at sbt.EvaluateSettings$INode.evaluate(INode.scala:135)
	at sbt.EvaluateSettings$$anonfun$sbt$EvaluateSettings$$submitEvaluate$1.apply$mcV$sp(INode.scala:69)
	at sbt.EvaluateSettings.sbt$EvaluateSettings$$run0(INode.scala:78)
	at sbt.EvaluateSettings$$anon$3.run(INode.scala:74)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1152)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:622)
	at java.lang.Thread.run(Thread.java:748)
[error] java.security.AccessControlException: access denied ("java.io.FilePermission" "/home/vinod/repos/freechips/rocket-chip/firrtl/src/test/resources/integration" "readlink")
[error] Use 'last' for the full log.
Project loading failed: (r)etry, (q)uit, (l)ast, or (i)gnore? 

I am not very acquainted with sbt/java/chisel/scala and hence I am not able to find what the error is actually. Does this have to do with some sbt or java packaging? Please help.

vinod1993, 2017-10-23:
For now, I reverted to the latest Travis CI build which passed and in that the build is passing (82b1aa8)

terpstra, 2017-10-23:
Whenever I get this error I have to blow away all the built files in my project. It's super annoying and I think an sbt bug.

closed 2017-11-07 by hcook