chipsalliance/rocket-chip #1067: Writing to user mode bits in MIE register

adich0, 2017-10-24:
Hello,

I'm attempting to set the UEIE bit in the MIE register, but am having trouble doing so. As a test, I wrote 0xFFF to the MIE register, but only notice the M*IE bits responding. I have set the useUser and useVM RocketCoreParams to true and false, respectively. Has someone seen a similar issue? I'm running off commit 8e1a002c4efab6360861ff0755b4fbd47d623c34

Thanks

mwachs5, 2017-10-24:
In order to have user mode interrupts, you need the RISC-V `N` extension, which I don't believe is currently supported by rocket-chip. `useUser` is necessary but not sufficient to have user-mode interrupts.

adich0, 2017-11-02:
On kind of a similar note, I'm having a bit of trouble determining if I'm actually in user mode.  I'm setting `mepc` to where I want to run user-mode code, clearing `mstatus.MPP`, then executing the `mret` instruction. I see the program counter jump to the `mepc` value, and there I try to read the `mtvec` CSR. I would expect some sort of exception to occur, but don't see one. Any ideas?

The following are the relevant code snippets. In my main C loop:
```

  // Switch to user mode and jump to 0x80006000
   temp = 0x80006000;
   __asm__ __volatile__(
      "csrw mepc, %0\n\t" : : "r"(temp)
   );     
   
   __asm__ __volatile__(
      "csrci mstatus, 12\n\t"
      "csrci mstatus, 11\n\t"
      "mret\n\t"
   );
```

And the instruction that I attempt to execute after jumping to the value in `mepc` :

```
/* Dummy user code */
  .section ".dummy_user","a"
  .align 2
  .global __dummy_user

__dummy_user:
   csrr sp, mtvec
   nop
   nop
   nop
   nop

  .size __dummy_user, .-__dummy_user

```

Any clues would be greatly appreciated.

245950258, 2019-05-08:
@adich0 I am very sorry for disturbing you. I try to work on the user mode. I want to know if the code you attempted to execute after jumping is in the main program. Could you show me full codes?

245950258, 2019-05-13:
@adich0 Have you ever encountered such a problem that I failed to set the UIE and UPIE bit in the mstatus register.

adich0, 2019-05-13:
@245950258 From a couple of posts above, I believe that the `N` extension is required for the user-mode CSRs.