chipsalliance/rocket-chip #1068: CSRFile does not support superscalar decode.

ccelio, 2017-10-24:
The CSRFile informs the pipeline of any illegal instructions by providing a `decode` interface (bits go in, illegal info goes out, can't explain that).

However, only a single decode port is provided, making it not possible to use the current rocket-chip CSRFile for superscalar processors.

https://github.com/freechipsproject/rocket-chip/blob/2786e42d994c504c1e54810187121ddad604882f/src/main/scala/rocket/CSR.scala#L174-L182

I'm not sure of the best way to tackle this. One easy solution is to make `decode` a Vec.

Is there a better way to provide a "function" that can be called against the CSRFile that provides a bit vector/tuple of illegal signals?

albert-magyar, 2017-10-24:
Is there some complexity that makes it break when catching illegal CSR accesses at the point where CSR accesses are serialized?

ccelio, 2017-10-24:
exceptions are caught in decode before being placed into the ROB. There are other types of illegal instructions it catches too -- FP illegal, ROCC illegal, etc. -- which depend on the state of the CSRFile.

aswaterman, 2017-10-24:
The decode port doesn't cause any state to be mutated, so I don't think there are any concerns about serialization.  Making it a Vec seems like the right way to go.

ccelio, 2017-10-25:
Here's my go at it:

https://github.com/freechipsproject/rocket-chip/commit/25ed3140dde91616f183ef13774f8e4d912462f2

 However, I have no idea what `decodeWidth` means in Rocket, especially since there's a `decodeWidth==2` when RVC is enabled, so I'm sure this creates 2x decode ports on the CSRFile for RVC cores.

aswaterman, 2017-10-25:
> 25ed314

Fine with me, you can PR it if you think this is the right way of dealing
with it for boom

PR #1069 (2017-10-25): csr: allow for superscalar decode
   * CSR provides a decode port to check for an illegal instruction.
   * This commit now allows for multiple instructions in decode to get this
      illegal instruction information.
   * This commit leverages the existing decodeWidth parameter. This will
      potentially over-provision the number of decode ports needed for
      RVC-enabled cores.

Closes #1068
commit c4978712 (2017-10-25): csr: allow for superscalar decode (#1069)

* CSR provides a decode port to check for an illegal instruction.
   * This commit now allows for multiple instructions in decode to get this
      illegal instruction information.
   * This commit leverages the existing decodeWidth parameter. This will
      potentially over-provision the number of decode ports needed for
      RVC-enabled cores.

Closes #1068
closed 2017-10-25 by ccelio