chipsalliance/rocket-chip #1070: How to configure rocket-chip with L2 Cache?

cclacc, 2017-10-26:
I just try to change the parameters nBankPerChannel, but cannot generate verilog with L2 successfully

Can anyone tell me how to configure rocket-chipwith L2 Cache?

terpstra, 2017-10-27:
Currently, rocket-chip only has a BroadcastHub for coherence.

closed 2017-11-07 by hcook

poweihuang17, 2018-01-08:
Interesting, I will take a look at BroadcastHub.

LoveZJT, 2019-05-22:
So, can rocket-chip support L2 cache now?

OrkunAliOzkan, 2023-07-25:
Hi @terpstra,

Is BroadcastHub L2 still supported or even up to date?

jerryz123, 2023-07-25:
There is documentation here: https://chipyard.readthedocs.io/en/stable/Customization/Memory-Hierarchy.html#the-sifive-l2-cache

The open source L2 is functional, but has not been further developed since when it was open sourced.

OrkunAliOzkan, 2023-07-25:
Hi @jerryz123,

First of all, thank you for taking the time to respond to my question. I really appreciate your willingness to help.

However, I feel that your answer didn't quite address the specific question I face. My question was related to whether `InclusiveCache` and in extension the cache generator from Chipsalliance has replaced the `Broadcast Hub` and in turn the SiFive inclusive cache generator, or is it such that both as equally supported as each over. I only ask this because the Chisalliance inclusive cache generator seems to have much more traction than the SiFive equivalent it seems to aim to replace.

jerryz123, 2023-07-25:
They are two different approaches to providing coherence, so they would be equally supported.

chipsalliance/rocket-chip-inclusive-cache is the same as sifive/blockinclusive-cache-sifive. Since the implementation has been donated to chipsalliance, the chipsalliance fork is being maintained, and the original sifive repo will likely not see any updates.

OrkunAliOzkan, 2023-08-25:
Hi @jerryz123 ,

Thank you for providing such timely support all of these times. It really helps me take on next steps when I get stuck.

I had an unrelated question to the topic above that I wanted to ask regarding the rocket-chip-inclusive-caches coherency protocol. I often find that documentation from sifive state that the coherency protocol used on tilelink is either MOESI or MESI adjacent, which is a bit conflicting, could you shine some more light into this? What is the coherency protocol used on the chipsalliance/rocket-chip-inclusive-cache? Is it configurable? Which module within the inclusive-cache handles maintaining coherency?