chipsalliance/rocket-chip #1087: Benchmark (Dhrystone) numbers mismatch?

jwang-us, 2017-11-03:
Hi,

I'm doing some research using RISC-V RocketChip. When I'm trying to compare the DMIPS/MHz, I found this presentation on HotChips https://www.hotchips.org/wp-content/uploads/hc_archives/hc27/HC27.25-Tuesday-Epub/HC27.25.70-Processors-Epub/HC27.25.731-RISC-V-Lee-UCB-V4.with-backup.pdf

It says on page 15, RocketChip's DMIPS/MHz is 1.72.

But when I'm running emulator (Verilator), the result is like
```
Dhrystones per Second:                                       1807
```
Therefore, according to the definition of DMIPS/MHz on wikipedia https://en.wikipedia.org/wiki/Dhrystone, DMIPS/MHz = 1807 / 1757 = 1.028

And when I'm using SPIKE to run Dhrystone, I got
```
Dhrystones per Second:                      1968
```
It means SPIKE's DMIPS/MHz is 1.12.

1. The number from the paper (presentation) is much higher than the one from Verilator. There must be something wrong, so please correct me if I'm wrong in my calculation.Thanks.
2. Or the number printed out should not be interpreted like this?  
2. SPIKE is a behavior function simulator, and RocketChip as a single-issue pipelined implementation, so RocketChip's score cannot be higher than SPIKE. Is that right?

Thanks a lot for your time.

Jim

aswaterman, 2017-11-03:
Your calculations are correct.  The dhrystone binary generated by the riscv-tests repository has low performance for a few reasons (like the linking model, and the lack of an optimized C standard library).

When using the Linux/glibc toolchain, I see slightly better performance than the Hot Chips presentation you cited on the current RTL.

aswaterman, 2017-11-03:
(And if you pretend that Spike is running at IPC=1, I observe that Spike gets 1.9 DMIPS/MHz, indicating that Rocket's IPC is a little over 0.9.)

jwang-us, 2017-11-04:
@aswaterman Thanks for your answer.

closed 2017-11-04 by jwang-us

IC-Dream, 2018-05-23:
@aswaterman, So, how could i use the Linux/glibc toolchain to run simulation, since i am a freshman~

zhuzhzh, 2020-11-22:
@phdbreak99 I want to know how you run riscv-test/Dhrystone on spike or verilator.

on my side, the dhrystone compiled with build.sh(in riscv-tools) can't be run with spike or verilator.

but other isa benchmark liks (riscv-tools/riscv-tests/build/isa/rv64ui-p-jalr) can be run on both platform.

is dhrystone needs glibc instead of newlib?